EdgeTX/edgetx #3227: UI 2.9 Button List Selection styling

JimB40, 2023-02-21:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Button List Selection detailed styling info
[image]

### Describe the solution you'd like

-

richardclli, 2023-05-06:
@JimB40 How about the "disabled" style?  I am working on the Flysky options UI and need to disable some controls when receiver is connected.  And I think in general, bind time options need to be disabled when receiver is connected.

JimB40, 2023-05-09:
My bad. :)
DISABLED is ENABLED with 50% opacity
[image]