EdgeTX/edgetx #3230: Flysky AFHDS2 Telemetry Gps issue

pierrotm777, 2023-02-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Hello,
I use a Radiomaster TX16S MK2 4in 1 transmitter.
I try to discover Gps coordinates but i receive lat lon with only two digit after the dot.
It's a know issue ?
How i can solve that please.
I have no issue with other protocols like Frsky, Hott or Multiplex.
Thanks for your help,

Pierre

### Expected Behavior

I should like to have the Gps coordinates with maximum digits.

### Steps To Reproduce

Discover telemety

### Version

2.7.1

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

...

doloebig, 2023-02-28:
GPS Speed, change the Units does not adapted the Value; show always m/s

pierrotm777, 2023-03-01:
Please, can you solve this gps issue.
See here same issue in OpenTx.
https://github.com/opentx/opentx/issues/6246

pierrotm777, 2023-03-25:
Do you think have a solution for the 2.9 release please?
Thanks in advance,
Pierre

pfeerick, 2023-03-25:
Need to check if https://github.com/EdgeTX/edgetx/pull/3386 fixes this, as this is from Flysky to fix a number of the telemetry issues specific to their system. Otherwise, https://github.com/opentx/opentx/pull/6252 is probaby needed here also.

pierrotm777, 2023-03-25:
Ok, thanks!