EdgeTX/edgetx #3235: Model Creation Issues

CorruptName, 2023-02-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If I create a new model it does not show until after reboot. If I copy a model the name is incorrect until after reboot.

### Expected Behavior

I expected the new model to be available without a reboot. I also expected the copy to have the correct name without a reboot.

### Steps To Reproduce

Create a new model
Model doesn’t show until reboot

Copy & Paste an existing model
Name isn’t correct until after reboot

### Version

2.8.0

### Transmitter

Radiomaster Zorro

pfeerick, 2023-02-22:
> If I create a new model it does not show until after reboot.

I'm not sure what is happening here - as soon as I create a new model, it shows in the model select list, and is set as the active model.

>  If I copy a model the name is incorrect until after reboot.

Reproducible - the model select list won't update with the name of the copied model until you either set the copied model as the active one, or power cycle the radio (at which point it scans all the model names again).

CorruptName, 2023-02-24:
> I'm not sure what is happening here - as soon as I create a new model, it shows in the model select list, and is set as the active model.
>
>
> Reproducible - the model select list won't update with the [...]

I have 2 of these Radiomaster Zorro TX. One is standard 4in1 the other is Pyrodrone ELRS
They were both purchased the same week, about 2 weeks ago. Both exhibit these issues and are both currently still on the stock 256mb SD Card. Perhaps I will try to use a different SD Card later.

Gerold68, 2023-02-26:
May the ussue I have posted is similar  X9E: edit issue on the radio #2951