EdgeTX/edgetx #3241: with the current "main f76aea3" no sound is played anymoreno sounds

ParkerEde, 2023-02-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

with the current "main f76aea3" no sound is played anymore. In addition, the entire simulator runs much slower than that of the previous "mains". Keystrokes such as Enter are not always recognised/executed.

### Expected Behavior

1. Sounds should play
2. the reaction speed should be the same as before.

### Steps To Reproduce

just use the simulator. I could reproduce it with X10S Express and TX16S

### Version

Other (Please specify below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS), Radiomaster TX16S / TX16SMK2

### Anything else?

main f76aea3

ParkerEde, 2023-02-22:
I use Windows Companion / Simulator

gagarinlg, 2023-02-22:
Did you install SDL2?
You can do this manually or by running the system specific prepare script.

ParkerEde, 2023-02-22:
No. Where I have to install SDL2, in Windows or in MSYS2 environment?

gagarinlg, 2023-02-22:
See this discord message https://discord.com/channels/839849772864503828/839856795781431317/1077911635571789895

rotorman, 2023-02-22:
Short version: run `pacman -S --noconfirm mingw-w64-x86_64-SDL2` in your MSYS2 and use -DSDL2_LIBRARY_PATH instead of -DSDL_LIBRARY_PATH (w/o 2 in the name) on the CMake line.

ParkerEde, 2023-02-22:
ah okay thank you very much. build is running...

raphaelcoeffic, 2023-02-22:
> Did you install SDL2?
> You can do this manually or by running the system specific prepare script.

Isn’t it distributed with Companion?

ParkerEde, 2023-02-22:
yes, now it runs as expected, and sound works too. 
Is it correct that the Companion is now just under 90MB in size instead of 25MB before?

raphaelcoeffic, 2023-02-22:
> yes, now it runs as expected, and sound works too. Is it correct that the Companion-Setup file is now just under 90MB in size instead of 25MB before?

The archive is around 30MB, which is about what we had lately. Did you happen to have compiled it yourself? If yes, it may be missing the stripping of debug information, which use a lot of space.

ParkerEde, 2023-02-22:
Yes, I build myself. How can I stop the debug information?
My selfbuilds were only 25MB before I made the changes mentioned above.

ParkerEde, 2023-02-22:
there are many lib*.dll files now in the installed Companion-folder.
Here a verfication 2.9 before and after
[after.txt](https://github.com/EdgeTX/edgetx/files/10805285/after.txt)
[before.txt](https://github.com/EdgeTX/edgetx/files/10805286/before.txt)

rotorman, 2023-02-22:
run 'strip <filename>'

ParkerEde, 2023-02-22:
I think companion.exe and simulator.exe are not the problem, they are still about the same size (about 20 MB) before and after. The problem is that there are many new lib*.dll files. If you tell me that is ok and these are now needed, then it is ok with me. 
I just want to avoid unnecessary DLLs being delivered and the setup becoming unnecessarily large.

ParkerEde, 2023-02-22:
OK. Thank you very much. Then I can close the ticket?

gagarinlg, 2023-02-22:
I think so

closed 2023-02-22 by ParkerEde