EdgeTX/edgetx #3242: no "telemetry lost" , RSSI low and critical sound warning

ParkerEde, 2023-02-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

Since "main f76aea3"
In the simulator, there is no longer a soundmessage that telemetry has been lost when "Telemetry Simulation" is unchecked.
If the corresponding RSSI threshold values are undershot, the sound files for low RSSI and critical RSSI are also not played.

### Expected Behavior

Soundmessages for telemetry warnings should be played

### Steps To Reproduce

1. start simulator
2. activate telemrty simulation
3. check "Simulation" and make sure the RSSI Value is working
4. uncheck "Simulation" = no "telemetry lost" soundmessage
5. check "Simulation" = no telemetry okay soundmessage
6. decrease the RSSI value in Telemetry Simulator to a smaller value of low and critical threshold = no soundmessages

### Version

Other (Please specify below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS), Radiomaster TX16S / TX16SMK2

### Anything else?

main f76aea3

ParkerEde, 2023-02-22:
Info: SDL2 is installed and other sound messages are working (hello.wav) and sounds triggered by switches.

ParkerEde, 2023-02-22:
in radio firmware X10S Express it works correct.

philmoz, 2023-02-22:
Do you get the 'Welcome to EdgeTX' sound on startup in the simulator?
This is no longer working for me.

pfeerick, 2023-02-23:
I'm getting the welcome audio (hello.wav) on with the latest build of companion - https://github.com/EdgeTX/edgetx/actions/runs/4241794073 - TX16S simulator. 

I also don't seem to be able to trigger the RSSI low/critical audio in the simulator, but if it's low or critical and I restart the simulator it plays then.

philmoz, 2023-02-23:
I rebuilt everything from scratch and the welcome sound came back in the simulator - vey strange.

However there seems to also be a problem on the radio as well.

I just tried connecting to some of my planes and was also not getting the telemetry sounds on the TX16S.
I scanned for new sensors and a new RSSI appeared with a different ID - 24 instead of 248.
When I deleted the old one (248) the sounds started working again.

philmoz, 2023-02-23:
The simulator issue is due to inverted logic from commit 72193eda8

closed 2023-02-23 by pfeerick

pfeerick, 2023-02-23:
> I just tried connecting to some of my planes and was also not getting the telemetry sounds on the TX16S. I scanned for new sensors and a new RSSI appeared with a different ID - 24 instead of 248. When I deleted the old one (248) the sounds started working again.

I'm seeing this also, but different IDs must be a different issue. the fact that it's 24 rather than 248 feels like a character is being dropped somewhere. In my test case with R168 FrskyX2 Rx, both RSSI and RxBt shifted, but TRSS and TQly are fine. Possibly due to something else out of that same PR.

PR #3246 (2023-02-23): fix(color) - Fix telemetry checks not running in simulator due to inverted logic.
Fixes #3242