EdgeTX/edgetx #3250: CRSF module not working with 400k baudrate.

JimB40, 2023-02-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When baudrade is set to 400k with CRSF protocol on external modul there is no communication with module.
- tested with mictoTX & Full size TX

### Expected Behavior

With same settings works seamlessly on 2.8.1

### Steps To Reproduce

- Set external module to CRSF
- set baudrate to 400k

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

Nightly build from 2023-02-23

pfeerick, 2023-02-24:
Can you be more specific - is this an actual Crossfire module, or something that talks CRSF (thus Crossfire, Tracer, ELRS)? 

I cannot reproduce this with TX16S and HM ES24TX, but can with TBS Tracer / Micro TX. I also noticed the `Errs` counter is missing, but that could be because it's not longer tracked?)

JimB40, 2023-02-24:
genuine TBS modules
- CRSF microTX
- CRSF FullTX
- Tracer microTX

raphaelcoeffic, 2023-02-24:
@pfeerick sorry for the error counter, I should re-add it again 🤭

ror1948, 2023-03-17:
Not working for me either using TBS crossfire external module on TX16s
no work at 400k but works at 115k
Tried on two different Matek Rx

ror1948, 2023-03-20:
same problem with nighty build and latest crossfire CRSF microTX V2 6.19
I have reverted back to TX16S edgetx 2.8.1
thanks

ror1948, 2023-03-23:
we are still having nightly releases? i thought releases are blocked until this bug is fixed ?

pfeerick, 2023-03-23:
There is no connection between a release and nightlies - nightlies are just that - nighly builds of `main`

pfeerick, 2023-03-29:
I've just tried the latest nightly since there have been a few PRs cleaning up serial timing and module handling, and Tracer _seems_ to be stable again at 400k on the TX16S, but not above or below that, so it seems the current state of affairs is there is a slight improvement, but still more to go.

ror1948, 2023-04-01:
yes 400k is working on crossfire .
But if I switch to 115k it stops working.
If I switch Tx back to 400k it still does not work.
I have to do a reboot of Tx to get it to work.
Also switching models and back also makes it work.
What is danger of losing link during flight?

pfeerick, 2023-04-02:
That would suggest something isn't resetting properly when the baud rate is changed on the fly, and either needs the module to be power cycled or only gets reset as part of that module off and on process  (since this happens when you switch models).

As long as you don't change baud rate during flight, shouldn't be an issue, but it is concerning that 115k doesn't appear to be working, even though its only supposed to be the fallback rate for older transmitters that cant reliably do 400k (per the TBS docs). Especially as 2.8 does not have all the changes to serial module handling that would explain why it would change in behaviour.

raphaelcoeffic, 2023-04-23:
@JimB40 @ror1948 I have some troubles reproducing anything here. For me it just works at 440k and 115k, and I can freely switch between the 2, each time trying out the TBS Agent Lite, and it just works and display proper signal. Now the question is: does the issue only show up if a receiver is connected? (I'd have to find one somewhere.....)

JimB40, 2023-04-23:
Tested. Work properly on 400k. On 115k TBS Agent do not get response (communiation module-radio is still active an works). but recommendation for TBS modules is 400k and 115 never worked 100% reliably.
@ror1948 can you check last nightly?

gagarinlg, 2023-05-06:
Is there any reason to keep this open? It looks to me like it is resolved.

pfeerick, 2023-05-07:
It hasn't been. ;)

pfeerick, 2023-06-02:
I've tried this again, with the latest nightly and TX16S, TBS Tracer Micro JR module and Nano RX hooked up to an iNav FC, and it seems to be working properly now. The control link always appears to be stable at 115k and 400k , but does not like anything above that (with module firmware v6.1.9). Given 115k is only "supported" for legacy radios that can't run a stable 400k baud rate, and that TBS Agent Lua is working properly at 400k (admittedly, occasionally laggy, but that is a different issue) - and even somewhat intermittently at 115k - I think this can be safely closed until some further reason for concern is raised.

closed 2023-06-02 by pfeerick