EdgeTX/edgetx #3252: Color LCD: SD card browser does not use the full width of the screen

ParkerEde, 2023-02-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On Color LCD Radios (landscape) like Horus etc the SD card browser does not use the full width of the screen. 

### Expected Behavior

The full width of the screen should be used, because then files with somewhat longer names could also still be displayed in single lines. This makes it look clearer

### Steps To Reproduce

open the SD-Card Browser on Radio or Simulator

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS), Radiomaster TX16S / TX16SMK2

gagarinlg, 2023-02-24:
The right side is used for image previews

ParkerEde, 2023-02-24:
would it not be a possibility to activate the preview for images only in the dedicated image folder? There the files have only 6 characters length, so it would fit well

philmoz, 2023-02-25:
Themes and scripts folders can have images with arbitrary filename lengths.

But I agree it is a waste of space at the moment - perhaps the preview should be a popup.

pfeerick, 2023-02-25:
It's not realistic to limit it to the IMAGES folder, as other folders may (and do) contain images, but I agree that it is poorly used screen real estate. I see some of the options (in no particular order) as being:
- make use of the space to show file information - filename (thus showing full filename if clipped), file size, other info if the file can be decoded (i.e. firmware bin might show version number, wave/mp3 would show duration)
- get rid of it, and instead have a full-screen image viewer
- have an option to toggle the preview panel on or off, period