EdgeTX/edgetx #3263: UI 2.9 Pop-up List styling

JimB40, 2023-02-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Pop-up List styling detailed info
[image]

### Describe the solution you'd like

-

ggoraa, 2023-02-26:
I would've made some corners more rounded tbh, to make it look more "polishes" or smth

Eldenroot, 2023-02-27:
Border-radius for popup?

ggoraa, 2023-02-27:
Yep, and also for the container that holds popup items