EdgeTX/edgetx #3264: Protocol Invalid - 2.8.1

cpwilson, 2023-02-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Since upgrading to 2.8.1 I am unable to bind to Xk X450 model using the correct MPM setting, I get the error message, Protocol Invalid om the protocol selection page.
Reverting back to 2.8.0 sees the issue resolved and the model binds fine again.      

MPM is running the same v1.3.3.20 firmware

### Expected Behavior

Would expect it to work as it does in 2.8.0

### Steps To Reproduce

Running 2.8.1
1 - select internal module protocol XK
2 - select sub protocol X450
3 - Unable to bind and get Protocol Invalid message

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

cpwilson, 2023-02-26:
2 images show internal RF Screens on each EdgeTX firmware version

2.8.1 shows 'Protocol Invalid'
2.8.0 shows 'V1.3.3.20 AETR'
[image]
[image]

richardclli, 2023-02-27:
Strange this should be fixed before 2.8.1 is release.  Let me check.

pfeerick, 2023-02-27:
No, this is also broken in 2.9 - and is probably related to a change in the MPM table Raphael made which I think was specific to the XK

richardclli, 2023-02-27:
Interesting, there are information specific to XK?  And affect XK alone?  Do you know the related PR/issue?

pfeerick, 2023-02-27:
Actually, no, it might have been a different protocol... Kyosho2 ... https://github.com/EdgeTX/edgetx/pull/3032

richardclli, 2023-03-19:
Figured out one error in the code modules_constants.h, correct code should be:

  MODULE_SUBTYPE_MULTI_LAST = MODULE_SUBTYPE_MULTI_KYOSHO2

After fixing this did not fix the XK problem, strange.

PR #3369 (2023-03-22): Fix MPM XK shows invalid protocol

Fixes #3264

Summary of changes:
closed 2023-03-22 by pfeerick