EdgeTX/edgetx #3267: sometimes the radio does not start (freeze in startscreen)

ParkerEde, 2023-02-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

sometimes the radio does not start. The radio remains frozen in the start screen. The power button and all other buttons then have no function. I have also waited more than 5 minutes, nothing happens. After I unplugged the battery and then plugged it back in, restarted the radio, the emergeny screen comes up immediately. Then the radio can be switched off regularly via the power button. When the radio is then restarted, it comes up normally.
I noticed that shortly after the start screen appears, the display gets a little darker. Then comes the startup sound (hello.wav) and it starts up normally. If the radio freezes during startup, the display in the startup screen remains at the initial brightness.
I have already tried different SD cards, it is also not. As soon as I go back to a slightly older 2.9.0nightly, I no longer have the problem.

### Expected Behavior

the radio should start reliably at all times

### Steps To Reproduce

1. Power on
2. Power off
3. Power on
4. Power off
5. somtimes up to 10 times

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

2.9.0 nightly 4fa666e6

ParkerEde, 2023-02-26:
Another hint: After I had to turn off the radio by disconnecting the battery, I made a backup from the SD card. On closer examination I noticed that the labels.yml is 0 bytes in size.
Another hint: After I had to turn off the radio by disconnecting the battery, I made a backup from the SD card. On closer examination I noticed that the labels.yml is 0 bytes large.
But when starting with this SD card, all models are assigned to the correct labels and the labels are all there again.

ParkerEde, 2023-02-27:
I updated my MSYS2 environment today with the setup_buildenv_msys2_stage1.sh and setup_buildenv_msys2_stage2.sh scripts and rebuilt the current main 4547f08c. I have done 50 radio starts so far and the problem doesn't seem to occur anymore. What do you think, could something have been wrong with the build environment that could have caused such a problem?

pfeerick, 2023-02-28:
Ah, so this was a self-build, not a github build? As that is important, as self-build can't be reproducibly tested like a GitHub build can be (since it's the same binary being used). 

Trying a different commit doesn't 100% mean the issue is gone, it is not impossible that it is an intermittent bug that comes and goes... rather than being 100% reproducible... and changes elsewhere in the code have obscured it. Or it *could* have been something in the build system that wasn't right after the recent SDL2 changes, etc. Need to minimise how many variables you change to eliminate that... i.e. since it occured with https://github.com/EdgeTX/edgetx/commit/4fa666e6807797fde13d6459ba068f0a00bd780b for you, did you try building that commit again? 

Having said that, there is nothing that has been committed lately that *specifically* relates to the startup code, really not sure what the culprit is. The worst sort of bug... the one you can't quite put your finger on.

mha1, 2023-02-28:
@pfeerick I can confirm this issue exactly as described by @ParkerEde. X10 Express with https://github.com/EdgeTX/edgetx/commit/4ea77ee749e148e656bd4b3e4b62d12a346adb44 nightly (https://github.com/EdgeTX/edgetx/suites/11197050093/artifacts/572438128 - x10-access-4ea77ee.bin).

Note: same nightly with tx16s-4ea77ee.bin works fine. Seems to be a X10 Express specific problem.

mha1, 2023-02-28:
@ParkerEde Unfortunately I don't have a X10 Express readily available, gotta go through a friend. Would you mind narrowing this down, i.e. doing a binary search between 7a12317 and 4fa66e? It's about 54 commits if I counted right. Shouldn't take more than 5 or 6 flashes to identify the turning point.

ParkerEde, 2023-02-28:
I had compiled the same version 2.9.0 nightly 4fa666e for my club mates (X12s, TX16S). They have not had this problem until now.
That's why I think it only affects the X10 Express. The active model, which should be loaded at start-up, has an RX4R with ACCESS firmware receiver bound to the internal ISRM.

mha1, 2023-02-28:
I believe you are right because the TX16s build is running without problems on my radio too but I don't think it's a specific ACCESS problem. We also saw the problem with internal ISRM OFF and external module PPM too.

ParkerEde, 2023-02-28:
Okay. Then that can also be ruled out. That's good to know. Thank you.

ParkerEde, 2023-02-28:
> Ah, so this was a self-build, not a github build? As that is important, as self-build can't be reproducibly tested like a GitHub build can be (since it's the same binary being used).
> 
> Trying a different commit doesn't 100% mean the issue is gone, it is not impossible that it is an intermittent bug that comes and goes... rather than being 100% reproducible... and changes elsewhere in the code have obscured it. Or it _could_ have been something in the build system that wasn't right after the recent SDL2 changes, etc. Need to minimise how many variables you change to eliminate that... i.e. since it occured with [4fa666e](https://github.com/EdgeTX/edgetx/commit/4fa666e6807797fde13d6459ba068f0a00bd780b) for you, did you try building that commit again?
> 
> Having said that, there is nothing that has been committed lately that _specifically_ relates to the startup code, really not sure what the culprit is. The worst sort of bug... the one you can't quite put your finger on.

I have now built the commit id 4fa666e again with the updated build environment and the error no longer occurs. So it must have been the build environment. The firmware files BIN have a different size (old vs. new build environment) and also differ in content. Normally, I have found that if I build one and the same commit several times, the BINs only differ in content by the time, but are the same size. Therefore, I am pretty sure that it was only due to the build environment. 
What do you think?

mha1, 2023-02-28:
Well, I had it with the gitbuilt nightly...

ParkerEde, 2023-02-28:
> Well, I had it with the gitbuilt nightly...

Jepp, that's strange.
Is it possible that the build environment on the server was not updated at the time the builds were created?

ParkerEde, 2023-02-28:
Did you try to update your MSYS Environment with the setup_buildenv_msys2_stage1.sh (violett icon) and setup_buildenv_msys2_stage2.sh (blue icon) scripts and build again? At this point I would be really interested to know if you get the same result as I do.

ParkerEde, 2023-02-28:
Okay sorry guys. I rejoiced too soon. Now I've actually got the problem again.
So everything back to the beginning. As I said, I used a selfbuild. But that obviously makes no difference to what mha1 reported.

ParkerEde, 2023-02-28:
I have now installed the last Commit ID e637b534 on the radio before the changeover to SDL2 and will now observe this over days.

gagarinlg, 2023-02-28:
There is no way the SDL2 change has an impact on the startup behavior of the radios.
The changes are minimal and only in code the is used for simu, simulator and companion.

ParkerEde, 2023-02-28:
Ah okay. Then we can at least find out whether the error was introduced by another PR before or after.

pfeerick, 2023-03-01:
>  Is it possible that the build environment on the server was not updated at the time the builds were created?

No, it is a new build environment every single run.

ParkerEde, 2023-03-01:
I have just had the error again that the Horus X10S Express did not start. So it must have come in between
commit e637b53 and commit 7a12317. 
There are many commits in between, do you have a suggestion which commit I should build and test next?

mha1, 2023-03-01:
Well, do a binary search. Take the commit halfway in between the two. If it exhibits the problem take the commit halfway between this and the older one, if not the one halfway between the newer one. And so on. Let n be the number of commits between the two you mentioned you'll only need to build Log2(n)+1 times to find the evil one.

ParkerEde, 2023-03-01:
yes, that is exactly how I have already started the tests.
The error also occurs with commit 1913d65 from 13.02.2023. I can narrow it down that far.

ParkerEde, 2023-03-02:
Hello everyone, I have now found out from which commit the error occurs. It is 72193ed

mha1, 2023-03-02:
oh boy, that's the blockbuster, thanks for your effort!

@raphaelcoeffic any hints?

richardclli, 2023-03-03:
The big serialAPI change, ah........

mha1, 2023-03-03:
Does the X10Express have a LED that turns green at pressing the power button and then shortly after turns blue?

ParkerEde, 2023-03-03:
no. The LED is blue as soon as the radio starts.

mha1, 2023-03-03:
do you hear the welcome to edgetx audio prior to freezing?

ParkerEde, 2023-03-03:
no. If the radio freezes during startup, the display in the startup screen remains at the initial brightness. No hello.wav will play

ParkerEde, 2023-03-03:
here a video with a successful startup because of the blue LED question

https://user-images.githubusercontent.com/15068446/222822015-fe35ec37-ef8c-4341-8c1d-258a6874ed6a.mp4

gagarinlg, 2023-03-03:
Does anyone who has this issue access to a debugger?

mha1, 2023-03-03:
debugger yes, x10express no (friend has one but doesn't want to fiddle with radio but experience same problem)

mha1, 2023-03-04:
For what it's worth: we had a situation were the radio got stuck at every attempt to power it up. Took out the SD Card and renamed RADIO and MODELS to force a clean start. Powered up the radio successfully, calibrated the sticks. Happily powering up after that. Deleted the newly generated RADIO and MODELS folders and renamed the backups back to RADIO and FOLDERS. Still powering up happily.

Just thinking out loud. If regenerating RADIO (I don't think it's a model.yml issue) changes behavior and as it seem to be non-volatile because it works with the old radio.yml after that this change must have survived a power cycle. What is surviving a power cycle? Is it just RTC data?

mha1, 2023-03-05:
what we know she far:
- only X10 express seems to be affected 
- it occurs moderately frequent
- no obvious pattern, seems random
- the major change to the serial/module API introdced this
- 2.9. on X10 express is not even usable for testing right now.

Anyone with ideas? @raphaelcoeffic, @pfeerick, @gagarinlg? Anyone?

Anyone out there owning a X10 Express, a suitable debugger and willing to open up the radio and have go at squashing this bug?

gagarinlg, 2023-03-05:
Sorry, I do not have one.
Someone with a debugger would be super helpful.

ParkerEde, 2023-03-06:
@mha1 and @gagarinlg where do you come from? I'm from north germany.

gagarinlg, 2023-03-06:
> @mha1 and @gagarinlg where do you come from? I'm from north germany.

send me a PM via discord, my nick there is "Malte (GMT+1)"

ParkerEde, 2023-03-06:
One item in the list of tasks that have been completed caught my eye:
"* Remove useless delays".
Is it possible that one too many delay were removed here that might be needed?

mha1, 2023-03-08:
The commit f1c2d31c3 you are referring to has two identical removals of 2 seconds RTOS_WAIT's, both in dedicated firmware flash code. So highly unlikely the cause for the hangs. 

-  /* wait 2s off */
-  watchdogSuspend(500 /*5s*/);
-  RTOS_WAIT_MS(2000);
-
+  watchdogSuspend(50 /*500ms*/);
   pulsesStart();

HThuren, 2023-03-19:
I have this bug and a express x10s, anything I can do to help (not to try every PR), logging or something.
The bug do not come regular, maybe after some days without use of the radio (?)
I use version compiled 17.03, which include every PR to that point.

ParkerEde, 2023-03-19:
Hi @HThuren ,
you can take the current main and then check out PR#3349. This fixes the error. At least I haven't had a freeze since this fix. 
If you don't want to do that you have to rewind to the commit d80adc6.

ParkerEde, 2023-03-19:
This Bug is fixed by https://github.com/EdgeTX/edgetx/pull/3349

HThuren, 2023-03-20:
> This Bug is fixed by #3349

Ok, I give it at try with 3349 (I believe I already have this PR, but will check).

ParkerEde, 2023-03-20:
The real error was only found and fixed yesterday. If you had already applied #3349 before, you would have to renew it.

HThuren, 2023-03-20:
I'm sorry, but where unable to download 3349

ParkerEde, 2023-03-20:
this will work. Try
`git branch -D test-3349
git checkout -b test-3349
git pull origin pull/3349/head
git rebase --skip`

HThuren, 2023-03-20:
$ git pull origin pull/3349/head
give
fatal: couldn't find remote ref pull/3349/head

ParkerEde, 2023-03-20:
I think the best would be to clone the main newly.
Or 
git fetch -a
git pull
and than
git branch -D test-3349
git checkout -b test-3349
git pull origin pull/3349/head
git rebase --skip

I just tried it and it works fine.

pfeerick, 2023-03-22:
If you install the GitHub `cli` tool, it's a lot easier to checkout PRs ;)
https://github.com/cli/cli

i.e. `gh pr checkout 3349`

closed 2023-03-26 by pfeerick