EdgeTX/edgetx #3269: Timer never announces "hours" when called from a special function

lightbulbjim, 2023-02-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When a count-up timer has elapsed for longer than an hour, triggering an announcement via a special function will announce the time in minutes, not hours and minutes.

For example, a time of 1h20 will be announced as "80 minutes".

Note:

* If a timer has the "minute" option enabled then the minutely callout does use hours and minutes as expected.
* I haven't tested count-down timers.

### Expected Behavior

A time under one hour should be announced in minutes, eg 20 minutes should be announced as "twenty minutes".

A time over one hours should be announced in hours and minutes, eg 1h20 should be "one hour and twenty minutes".

### Steps To Reproduce

1. Configure a count-up timer starting at 00:00.
2. Configure a special function to call out the timer value when a switch is moved.
3. Let the timer count until more than an hour has elapsed.
4. Move the switch to trigger a time announcement.

### Version

Other (Please specify below)

### Transmitter

Radiomaster Zorro

### Anything else?

Running EdgeTX 2.8.1.

philmoz, 2023-03-03:
There is a special case in the code purely for special functions.

If the value is > 10 minutes then it plays as minutes, and only plays the minutes (no seconds). Under 10 minutes it plays both minutes and seconds.

I have no idea why this is the case; but I would suggest removing this and playing the time values consistently.

lightbulbjim, 2023-03-03:
> If the value is > 10 minutes then it plays as minutes, and only plays the minutes (no seconds).

I believe the thinking with that was that above a certain amount of time you don't care about seconds. I'm ok with that, but it sounds like times over an hour weren't considered when making that change.

PR #3293 (2023-04-19): fix(radio) - Announce timer value as hours and minutes when played from special function.
Fixes #3269
closed 2023-04-19 by pfeerick