EdgeTX/edgetx #3273: Companion X10s Express, ISRM receiver name missing

mha1, 2023-02-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Attached .yml was generated with a X10S Express (int. ISRM) using 2.8.1. It has one receiver listed and named. 

```
moduleData: 
   0:
      type: TYPE_ISRM_PXX2
      subType: ACCESS
      channelsStart: 0
      channelsCount: 8
      failsafeMode: CUSTOM
      mod: 
         pxx2: 
            receivers: 0
            racingMode: 0
            receiverName: 
               0:
                  val: "RX4R    "
failsafeChannels: 
   1:
      val: -348
   2:
      val: -41
```

After importing the model file with Companion 2.8.1 (settings X10s Express) no receiver name is listed. Saving the file from Companion results in .yml with no receiver name.

```
moduleData:
  0:
    type: TYPE_ISRM_PXX2
    subType: ACCESS
    channelsStart: 0
    channelsCount: 8
    failsafeMode: CUSTOM
    mod:
      pxx2:
        receivers: 0
        racingMode: 0
failsafeChannels:
  1:
    val: -348
  2:
    val: -41
```

[model1.yml.txt](https://github.com/EdgeTX/edgetx/files/10838053/model1.yml.txt)

[image]

### Expected Behavior

Receiver name is listed and can be edited in Companion
Saving the imported model.yml should not delete receiver names

### Steps To Reproduce

1. Set Companion to X10s Express
2. Import model.yml
3. Check Setup tab for receiver name
4. Save model.yml
5. Check model.yml for receiver name tag

### Version

2.8.0

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Anything else?

FW and Companion version 2.8.1

mha1, 2023-02-27:
@elecpower Care to check this out? Thanks - Michael

elecpower, 2023-02-27:
moduleData: 
         0:
            type: TYPE_ISRM_PXX2
            subType: ACCESS
            channelsStart: 0
            channelsCount: 8
            failsafeMode: CUSTOM
            mod: 
               pxx2: 
                  receivers: 0
                  racingMode: 0
                  receiverName: 
                     0:
                        val: "RX4R    "

The entry 'receivers: 0' is why Companion does not load the receiver name. Need to check if fault on Companion or radio side.

elecpower, 2023-02-27:
This needs some advice from the radio gurus as my investigations so far show

X10 Express libsim and radio is always saving zero against receivers label

      mod: 
         pxx2: 
            receivers: 0
            racingMode: 0
            receiverName: 
               0:
                  val: "SimuRX1"
               1:
                  val: "SimuRX2"

X9D+ 2019 libsim and radio is saving (I assume the max number of receivers ie 3)

      mod: 
         pxx2: 
            receivers: 3
            racingMode: 0
            receiverName: 
               0:
                  val: "SimuRX1"
               1:
                  val: "SimuRX2"

mha1, 2023-02-27:
@elecpower Neil, thanks for looking into this. Sure looks like a radio fw problem, and even looks like a x10s Express specific problem.

ParkerEde, 2023-02-27:
I have also looked at it with 2.9.0 main. Companion and X10S Express. The name of the receiver is correct in the yml file and the Companion shows it correctly.
Is it possible that the problem was already solved in 2.9.0?

pfeerick, 2023-02-28:
I thing there were some X10-specific / ACCESS RX changes, so quite possible.

mha1, 2023-02-28:
@pfeerick I browsed through the commits down to Dec 2022 but couldn't find anything related. Have an idea?

@ParkerEde Totally not related to this issue: as you seem to have a X10 express, can you please do me a favor, flash the 4ea77ee nightly (https://github.com/EdgeTX/edgetx/suites/11197050093/artifacts/572438128 - x10-access-4ea77ee.bin) and give it a few starts and power downs. Does it have problems starting up?

ParkerEde, 2023-02-28:
@mha1 did you mean this?:
https://github.com/EdgeTX/edgetx/issues/3267
Yes, I had problems with that, but it was a selfbuild. Interesting that you have it with a gitbuild as well. I think you should post your experiences there too.

pfeerick, 2023-03-01:
I would have expected no.3 - since `PXX2_MAX_RECEIVERS_PER_MODULE` is the MAX, and there is no point having entries for non-existent RXs.

pfeerick, 2023-03-01:
Look like it's used to determine and manage if the receiver slot is in use for B&W displays.

https://github.com/EdgeTX/edgetx/blob/4547f08cca75588576896c65cda15b33fb37b2c3/radio/src/pulses/modules_helpers.h#L735-L738

https://github.com/EdgeTX/edgetx/blob/4547f08cca75588576896c65cda15b33fb37b2c3/radio/src/pulses/modules_helpers.h#L740-L743

https://github.com/EdgeTX/edgetx/blob/4547f08cca75588576896c65cda15b33fb37b2c3/radio/src/pulses/modules_helpers.h#L750-L755

mha1, 2023-03-01:
This is odd. setPXX2ReceiverUsed doesn't set storage dirty and .receivers is used as a bitfield (expect numbers 0,1,3,7). 

If setPXX2ReceiverUsed is used for color radios too, the missing storage dirty would explain a permanent receivers: 0 in model.yml's and Companion not reading the receivers.

mha1, 2023-03-11:
any news about this?

closed 2023-03-23 by pfeerick