EdgeTX/edgetx #3274: The Frsky telemetry does not work anymore with Nightly.

Salto42, 2023-02-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The Frsky telemetry does not work anymore.
When searching for new sensors it works with new ID numbers.
The ID numbers have changed.
RBSMIX from Real Tadango does not work.

RBSMIX : https://github.com/RealTadango/FrSky

### Expected Behavior

That the Frsky telemetry works without modification.

### Steps To Reproduce

The last Nightly that still works is from 05/02/23.
The last test was done with the last Nightly

### Version

Nightly (Please give date/commit below)

### Transmitter

Jumper T16, Radiomaster TX16S / TX16SMK2

philmoz, 2023-02-28:
I'm seeing this as well - the change appears to be from commit 72193eda8 (New module API).

For example, before this commit RSSI id was 248 (0xF8)
The new id is 24 (0x18)

The difference is the top 3 bits are no longer set on the ID.
This corresponds to the frskyinstance.rxIndex field in the TelemetrySensor struct.

Salto42, 2023-03-18:
OK to change the ID number of the telemetry sensors, but the sensor "RBS" cannot be read in "RBSMIX.LUA" with "getValue("RBS")" nor in the logical switches.

Salto42, 2023-03-23:
Problem solved by changing the ID number and RBSMIX.LUA file

closed 2023-03-23 by Salto42

ischicopter, 2024-02-29:
> Problem solved by changing the ID number and RBSMIX.LUA file

Hello Salto, I am having the same issue with the RBSMIX.LUA file. Could you kindly go into more detail as to what changes you did exactly? Your help will be greatly appreciated, thank you!

kind regards

Salto42, 2024-03-01:
--
-- Copyright (C) OpenTX
--
-- Based on code named
--   th9x - http://code.google.com/p/th9x
--   er9x - http://code.google.com/p/er9x
--   gruvin9x - http://code.google.com/p/gruvin9x
--
-- License GPLv2: http://www.gnu.org/licenses/gpl-2.0.html
--
-- This program is free software; you can redistribute it and/or modify
-- it under the terms of the GNU General Public License version 2 as
-- published by the Free Software Foundation.
--
-- This program is distributed in the hope that it will be useful,
-- but WITHOUT ANY WARRANTY; without even the implied warranty of
-- MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
-- GNU General Public License for more details.
--

-- LUA Mix Script

local prevStatus

local prevRx1FrameLoss = false
local prevRx2FrameLoss = false
local prevRx1Failsafe = false
local prevRx2Failsafe = false

local rx1FrameLossCount = 0
local rx2FrameLossCount = 0
local rx1FailsafeCount = 0
local rx2FailsafeCount = 0

local function hasbit(x,p)
 if x == p then
 return (1)
 else
 return (0)
 end
end

--  "Rx1 Ovl", "Rx2 Ovl", "SBUS Ovl", "Rx1 FS", "Rx1 LF", "Rx2 FS", "Rx2 LF", "Rx1 Lost", "Rx2 Lost", "Rx1 NS", "Rx2 NS"

local function run()
  local status = getValue("RBS")  
  
  setTelemetryValue(0xb20, 0, 2, hasbit(status, ("Rx1 FS")), 0, 0, "R1FS")
  setTelemetryValue(0xb20, 1, 2, hasbit(status, ("Rx2 FS")), 0, 0, "R2FS")
  setTelemetryValue(0xb20, 2, 2, hasbit(status, ("Rx1 LF")), 0, 0, "R1LF")
  setTelemetryValue(0xb20, 3, 2, hasbit(status, ("Rx2 LF")), 0, 0, "R2LF")

  if hasbit(status, ("Rx1 FS")) == 1 then
	if prevRx1Failsafe == false then
		rx1FailsafeCount = rx1FailsafeCount + 1
	end
	prevRx1Failsafe = true
  else 
	prevRx1Failsafe = false
  end

  if hasbit(status, ("Rx2 FS")) == 1 then
	if prevRx2Failsafe == false then
		rx2FailsafeCount = rx2FailsafeCount + 1
	end
	prevRx2Failsafe = true
  else 
	prevRx2Failsafe = false
  end

  if hasbit(status, ("Rx1 LF")) == 1 then
	if prevRx1FrameLoss == false then
		rx1FrameLossCount = rx1FrameLossCount + 1
	end
	prevRx1FrameLoss = true
  else 
	prevRx1FrameLoss = false
  end

  if hasbit(status, ("Rx2 LF")) == 1 then
	if prevRx2FrameLoss == false then
		rx2FrameLossCount = rx2FrameLossCount + 1
	end
	prevRx2FrameLoss = true
  else 
	prevRx2FrameLoss = false
  end
  
  setTelemetryValue(0xb20, 4, 2, rx1FrameLossCount, 0, 0, "R1LC")
  setTelemetryValue(0xb20, 5, 2, rx2FrameLossCount, 0, 0, "R2LC")
  setTelemetryValue(0xb20, 6, 2, rx1FailsafeCount, 0, 0, "R1HC")
  setTelemetryValue(0xb20, 7, 2, rx2FailsafeCount, 0, 0, "R2HC")
  
  if status ~= 0 then
    return 10
  else
    return  0
  end
end

return { run=run }

ischicopter, 2024-03-05:
thanks! I believe there is also some compatibilty issue with my Radiomaster TX16s running EdgeTX. I found that rather looking for the specific bits in the telemetry, the code has to look for the actual "text" (Rx2FS, Rx1 LF etc.) in the telemetry section. So I came up with this using ChatGPT for a couple hours. To be clear, I have no coding experience at all . But for those who are interested:

I removed the R1FS sensor, as my RB-10 has only one S.Port and I am running the ACCST protocol, so telemetry on only one rx anway. I also added an Overload Counter which counts how often the "RBCS" sensor goes to anything else than "OK".

-- LUA Mix Script

local outputs = { "RFS" }

local prevRx1FrameLoss = false
local prevRx2FrameLoss = false
local prevRx2Failsafe = false
local prevOvld = false

local rx1FrameLossCount = 0
local rx2FrameLossCount = 0
local rx2FailsafeCount = 0
local OvlC = 0  -- Initialize OvlC count

-- Function to check telemetry status and update virtual sensors
local function checkTelemetry(status, rbcs, rb1v, rb2v)
  if status == "Rx2 FS" then
    setTelemetryValue(0xb20, 0, 2, 1, 0, 0, "R2FS")
    if not prevRx2Failsafe then
      rx2FailsafeCount = rx2FailsafeCount + 1
    end
    prevRx2Failsafe = true
  else
    setTelemetryValue(0xb20, 0, 2, 0, 0, 0, "R2FS")
    prevRx2Failsafe = false
  end

  if status == "Rx1 LF" then
    setTelemetryValue(0xb20, 1, 2, 1, 0, 0, "R1FL")
    if not prevRx1FrameLoss then
      rx1FrameLossCount = rx1FrameLossCount + 1
    end
    prevRx1FrameLoss = true
  else
    setTelemetryValue(0xb20, 1, 2, 0, 0, 0, "R1FL")
    prevRx1FrameLoss = false
  end

  if status == "Rx2 LF" then
    setTelemetryValue(0xb20, 2, 2, 1, 0, 0, "R2FL")
    if not prevRx2FrameLoss then
      rx2FrameLossCount = rx2FrameLossCount + 1
    end
    prevRx2FrameLoss = true
  else
    setTelemetryValue(0xb20, 2, 2, 0, 0, 0, "R2FL")
    prevRx2FrameLoss = false
  end

  -- Set Ovld based on RBCS sensor
  if rbcs ~= "OK" then
    setTelemetryValue(0xb20, 3, 2, 1, 0, 0, "Ovld") -- Set Ovld to 1 if not OK
    if not prevOvld and (rb1v > 1 or rb2v > 1) then
      OvlC = OvlC + 1  -- Increment OvlC count
    end
    prevOvld = true
  else
    setTelemetryValue(0xb20, 3, 2, 0, 0, 0, "Ovld") -- Set Ovld to 0 if OK
    prevOvld = false
  end
end

-- Function to run telemetry checks
local function run()
  -- Get telemetry data from "RBS", "RBCS", "RB1V", and "RB2V" sensors
  local rbsStatus = getValue("RBS")
  local rbcsStatus = getValue("RBCS")
  local rb1vStatus = getValue("RB1V")
  local rb2vStatus = getValue("RB2V")

  -- Check telemetry status and update virtual sensors
  checkTelemetry(rbsStatus, rbcsStatus, rb1vStatus, rb2vStatus)

  -- Set telemetry values for counting sensors
  setTelemetryValue(0xb20, 4, 2, rx1FrameLossCount, 0, 0, "R1LC")
  setTelemetryValue(0xb20, 5, 2, rx2FrameLossCount, 0, 0, "R2LC")
  setTelemetryValue(0xb20, 6, 2, OvlC, 0, 0, "OvlC") -- Set OvlC value
  setTelemetryValue(0xb20, 7, 2, rx2FailsafeCount, 0, 0, "R2HC") -- Set R2HC value

  -- Return appropriate value
  if rbsStatus ~= 0 then
    return 10
  else
    return 0
  end
end

-- Return function and output
return { run = run, output = outputs }