EdgeTX/edgetx #3275: CRSF V3 Crossfire/tracer Modules on higher baudrates

kamik2, 2023-02-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

when the module bay baud rate setting is changed to something other than 400kbau, the communication with the module breaks up.

My radio is modded ("frsky mod")  so there should be no hardware-related issue.

I'll try to get a capture of the communication by tomorrow.

it was tested with crossfire fw 6.10..6.19 and the actual alpha (not public)

### Expected Behavior

.

### Steps To Reproduce

- connect a crossfire/ tracer module to the radio module bay
- set the baud rate to 400kbaud 
- the communication should work fine
- change the baud rate to something higher 
- the communication breaks up

### Version

2.8.0

### Transmitter

FrSky X9D+

pfeerick, 2023-02-28:
What version of Tracer fw introduces supported baud rates higher than 400k?

kamik2, 2023-03-02:
sry, that was my fault. i was under the assumption that the higher baud rates are supported by the tx module as well.  but rapahel corrected me. sry