EdgeTX/edgetx #3277: attempt to yield across a c-call boundary error

lshems, 2023-02-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When running LUA scripts that run fine on OpenTX, they crash on edge 2.8.1 with the above mention

### Expected Behavior

No crash

### Steps To Reproduce

I don't have a real way of reproducing it. My scripts are more a set of functions in files.

I do not expect any fix. I just wanted to make sure that it is known that I have encountered this issue. So if any body else also finds it, with perhaps a reproducable case, it could serve as a more useful case ot try and solve the issue.

For now, I will not move to egde.

### Version

2.8.0

### Transmitter

Jumper T16

lshems, 2023-02-28:
It is not present in 2.7 !!

So i'll downgrade and keep there untill it is solved.

:)

lshems, 2023-03-02:
How do you test/debug lua code on the actual radio?

It would help if I could quickly test changed code, so I could eliminate some causes?

JimB40, 2023-03-03:
Yes I've encounter this error rarely too. Tried to find what is the reason but no luck so far.
My clues are that script is "too heavy" to be executed in one cycle then in LUA code there is function that causes this error. Last time I've observed it was when I tried to sort quite a large table with table.sort

What I found is:
_The error is telling you that you are attempting to yield from within Lua code where there is some C function between the Lua code doing the yielding and the Lua code that resumed the coroutine. To hit this error, what you have to have done is call some C function from Lua, which calls back into Lua code, which then calls coroutine.yield()._ 

Does it happen when you run script in one-time mode or as a widget?

Maybe @raphaelcoeffic or @jfrickmann will have some more clues checking where this coroutine.yeld() is invoked.

lshems, 2023-03-03:
It seems to be introduced when going to 2.8.

My scripts are essentially telemetry scripts, that are then loaded and processed in loader scripts for one-time scripts, widgets, ...

lshems, 2023-03-03:
So all crash on this. I think it is perhaps linked to the limitation of the background functions of widgets not being called (due to memory issues).

In some searches I did on Google a resume or yield when running several processes might cause this. I didn't understand much, but I taught it was worth mentioning.

lshems, 2023-03-03:
So we think in the same direction...

lshems, 2023-03-03:
It's a pity, because 2.8 is a huge improvement in terms of GUI

jfrickmann, 2023-03-03:
Does it crash on one-time scripts, widgets or both?

lshems, 2023-03-03:
Both

jfrickmann, 2023-03-03:
So it's not the preemption thing then, as widgets do not have that