EdgeTX/edgetx #3290: DSM GPS telemetry not working

lionelkeeping, 2023-03-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The telemetry from the Spektrum SPMA95871 GPS sensor attached to the AR637TA smart receiver is not understood by EdgeTX.  The sensor is recognized and new items are added to the telemetry data in the transmitter (SAT, Gspd and Date) but the data is incorrect.  SAT is a number about 162 (saw 9 and 320 once but generally between 162  - 167).  Gspd is zero but when I move the plane, the number jumps around erratically and is not a realistic speed.  Date flashes numbers but they are not dates.  The GPS also overloads the maximum of 40 telemetry slots. 

### Expected Behavior

I expected the radio to show me GPS positions, ground speed, and heading as it would on a Spektrum radio. I could  then show this on the telemetry screens.

### Steps To Reproduce

1. Connect the GPS to the SBUS on the receiver
2. Power up the transmitter 
3. Power up the receiver
4. In Telemetry screen, select Discover new
5. Radio will overload the telemetry slots
6.  Shut down the radio and restart to stop the overload
7. Examine the Gspd, Sat, and Date Telemetry lines.

### Version

Other (Please specify below)

### Transmitter

Radiomaster Zorro

### Anything else?

EdgeTX 2.8.1
4in1 firmware 1.3.3.20 AETR

frankiearzu, 2023-03-03:
I saw someone reporting this in Facebook.  Don't have the sensor to test, but have been in that part of the code recently.   Can you log the sensor data you are seeing and add it to this ticket.  Here is how to setup telementry data log:  https://www.youtube.com/watch?v=SsbnONkErbc.  Also please tell me what sensor names are you looking for.

lionelkeeping, 2023-03-03:
Hi there!  Thanks for responding.  It may have been the FB post I made that you saw.  I'll make some time over the weekend to set up the logging and make some tests and hopefully it'll be enough for you to make some sense of it.  It would be great to get this working!

Lionel

⁣Get BlueMail for Android ​

frankiearzu, 2023-03-04:
Lionel,
Im in a work trip in Germany for next week, so will not be able to work in this until then..  I added a feature on the DSM telemetry firmware (for 2.9.0 and already in nightly build) who exposes the RAW Spektrum telementry data to LUA scripts.    What i can do when i return is create a small LUA script to record your GPS RAW sensor data into a file.  In that way, you can send me the file and I can see what your sensor is sending.  After that will be much easier to correct the firmware code.  The feature was intended to support TextGen LUA scripts (example, to program AVIAN ESCs), but can also be used  to capture other sensor telemetry messages RAW data
.

lionelkeeping, 2023-03-04:
Sounds good.  I look forward to hearing from you!  Thank-you.  Enjoy your time in Germany.

⁣

frankiearzu, 2023-03-14:
Hi Lionel,
Created a DSM RAW telemetry capture tool to see what are you getting from your GPS. Just move the Lua script to your **/SCRIPTS/TOOLS** directory.  (unzip the attached ZIP file in your computer, and copy the .lua script).

After you run it, you will have a menu to capture some telemetry streams for GPS:
We want to capture GPS STAT (16), GPS LOG (17) and maybe GPS Binary (26).  
Quality of Serivce (7F) always work for Spektrum RX.

Run Quality of Service first for a little just to make sure your have the proper firmware.. if it is not capturing any lines, you don't have the right firmware  (see firmware description at the end of this message).

The log is written to: **/LOGS/dsm_raw_tel.txt**.  Is completly text so you can view it with Notepad or any other text editor.
The log file looks like:

`---------Capturing  I2C_ID 7F`
`7F 00  00  08 FF FF FF FF FF FF  00  00  00  00  02 99 20 AA`

It will continue adding to the log until you delete the file..

You will need one of the latest nightly builds of EdgeTX or you can download a 2.8.1+TextGen that i compiled for myself here:
https://github.com/frankiearzu/DSMTools/tree/main/EdgeTx_Firmware
I have been using that for flying, but you can easilly rollback the firmaware back and forth between official 2.8.0, 2.8.1 and my version.  The mechanism to capture this telemetry is the same that i am using to capture TextGen telemetry (ex, to program AVIAN ESC from your TX).

Once you have the file, send it here.

[](url)
[DSM_Capture_Tel.zip](https://github.com/EdgeTX/edgetx/files/10963496/DSM_Capture_Tel.zip)

lionelkeeping, 2023-03-14:
Perfect!  I'll have a look at it in the morning.  Thank you!

⁣

frankiearzu, 2023-03-14:
Thank you Lionel, 
Thats what i was expecting..  in the QoS, we are not displaying the quality of the signal, but how many messages has we received, so its expected to only increase.
I see data for message ID=0x16, that is GPS_LOG from the documetation.   Looks like no data on 26 (GPS_Binary), and i don't see any capture for message id=0x17 that is GPS_Status.. but i think the file here is truncated.

I have what i need to start, thank you, will check data agains the documentation and code.

Thank you,
Frankie

frankiearzu, 2023-03-14:
Lionel,
The file was truncated... can you try caputing message 0x17..  16 and 17 are the two messages that GPS generates. Ones is location, the other status.

Frankie

lionelkeeping, 2023-03-14:
Just to clarify, you want me to modify the script to capture 0x17?  I can probably manage it looking through what you sent but but I haven’t done that before.   I’m not at home now so I’ll have a look in the morning.  Thanks for getting back to me so soon!

From: Frankie Arzu
Sent: Tuesday, March 14, 2023 4:08 PM
To: EdgeTX/edgetx
Cc: lionelkeeping; Author
Subject: Re: [EdgeTX/edgetx] DSM GPS telemetry not working (Issue #3290)

Lionel,
The file was truncated... can you try caputing message 0x17.. 16 and 17 are the two messages that GPS generates. Ones is location, the other status.
Frankie

frankiearzu, 2023-03-14:
trying to make sence of the data..
Where are you located?? and aproximate what altitude?

This is the data of one of the messages, but just like that, it does not a valid coordinate.
Altitute-Low = 17 00      = -170.0m  or 17m
Latitude = 96 73 12 28    = 96.731228   or 28.127396 
Longitude= 08 72 09 82    = 08.720982   or 82.090872 
Course= 86 08 		  = 860.8 or 088.6
HDOP = 13 
GPSFlags = B9 = 1100 1001 = North, WEST, GPS_FIX_VALID, Negative Altitude

frankiearzu, 2023-03-14:
No, there should be an option in the menus to capture 16,17, 26 and 7F.   Let me try with a smaller screen.. i did my test on a TX16s.  Maybe is hiding the menu option for 17

lionelkeeping, 2023-03-14:
I'm at Zephyrhills, FL, USA so your 28.127 N and 82.0908 W would be correct.  That's awesome!  And elevation is pretty low in FL so 17m is about right.  And the course would have been close to 90 in one direction, then a short bit of about 180 and 90 in the other direction. 

You're pretty amazing with that!

⁣Get BlueMail for Android ​

frankiearzu, 2023-03-14:
Great!!, now i know how the data is encoded.  
Let me modify the scrip to show the GPS coordinates in the  screen for the messages.
Will also check the code in EdgeTx to see if it is processing the messages correctly.

frankiearzu, 2023-03-14:
I see why is not capuring the GPS_Status(0x17),, typo inside my script.. both menu options are capturing only message 16 (GPS_LOG)
Attached is new version of the screen.  this will properly capture message 17 (GPS_Status).

That will give us the date and number of satelites.

[DSM_Capture_Tel.zip](https://github.com/EdgeTX/edgetx/files/10974448/DSM_Capture_Tel.zip)

lionelkeeping, 2023-03-15:
Ahh, OK.  Happens to all of us!  It’s dark here now but I’ll add the new script and re-run tomorrow and send the data ASAP.  Thank you very much!

Lionel

From: Frankie Arzu
Sent: Tuesday, March 14, 2023 6:37 PM
To: EdgeTX/edgetx
Cc: lionelkeeping; Author
Subject: Re: [EdgeTX/edgetx] DSM GPS telemetry not working (Issue #3290)

I see why is not capuring the GPS_Status(0x17),, typo inside my script.. both menu options are capturing only message 16 (GPS_LOG)
Attached is new version of the screen. this will properly capture message 17 (GPS_Status).
That will give us the date and number of satelites.
DSM_Capture_Tel.zip

frankiearzu, 2023-03-15:
Based on what i learned today.. created a version of the firware from 2.8.1 with the GPSFix.

Tested on the TX16 and my FrSky QX7..  at least everything works as before. but i really can't test if the code that i added to fix the GPS will work, since i don't have a sensor.  Give it a try, but don't fly with it... just use it in the ground just to play safe.
You will need to discover sensors.  
Added "Sats" (number of Sat) ,"Date"(really is only time in HHMMSS), "GAlt" (Altitude),"GSpd", "GPS" (Latitude, Longitude) 

[Edgetx-Zorro-2.8.1-GPSFix.bin](https://github.com/frankiearzu/DSMTools/blob/main/EdgeTx_Firmware/Edgetx-Zorro-2.8.1-GPSFix.bin)

lionelkeeping, 2023-03-15:
Hi Frankie.  Attached is the new raw telemetry file.  This was done just walking the plane around on my property so it’s not a lot of data.

 

I tried to attach some video of what the screen shows on my Zorro as the plane is moved around but your mail server rejected it due to size.  I will try sending again in multiple emails.  If you need more, or something specific, let me know.

 

I think based on my observations, the ground speed may be recording properly but is off by a factor of 1000.  As the video shows, walking speed of about 0.8m/s is showing as over 800 m/s.  No idea if that’s correct but it looks reasonable.  I can’t make much sense of the other telemetry that’s displayed.

 

Let me know what else you need!

 

--------- Capturing I2C_ID 16
16 00 42 01 72 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 41 01 72 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 41 01 72 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 41 01 76 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 40 01 76 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 40 01 76 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 39 01 76 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 38 01 76 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 38 01 76 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 38 01 76 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 37 01 76 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 36 01 76 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 36 01 76 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 36 01 76 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 35 01 76 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 35 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 35 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 35 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 34 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 34 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 34 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 34 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 34 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 34 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 34 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 34 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 34 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 34 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 34 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 34 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 34 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 33 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 33 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 33 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 33 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 33 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 33 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 33 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 33 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 33 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 33 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 33 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 33 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 32 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 32 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 32 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 32 01 78 70 12 28 80 80 09 82 00 00 27 39 20 00
16 00 33 01 76 70 12 28 80 80 09 82 11 01 27 39 20 00
16 00 33 01 76 70 12 28 80 80 09 82 01 03 27 39 20 00
16 00 34 01 76 70 12 28 80 80 09 82 17 05 27 39 20 00
16 00 34 01 76 70 12 28 80 80 09 82 69 06 27 39 20 00
16 00 34 01 76 70 12 28 80 80 09 82 75 07 27 39 20 00
16 00 34 01 76 70 12 28 80 80 09 82 00 08 27 39 20 00
[... 654 lines cut ...]
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 60 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00
7F 00 00 04 FF FF FF FF FF FF 00 04 00 00 02 61 20 00

lionelkeeping, 2023-03-15:
Video attached

lionelkeeping, 2023-03-15:
Unfortunately, the video is rejected due to size limits by your mail server.  Let me know if you have an alternate address that will work.

 

Lionel

pfeerick, 2023-03-16:
Github does not accept email attachments. You might want to consider a service such as https://wetransfer.com if you want to send large files.

lionelkeeping, 2023-03-16:
Thanks Peter.  I sent the files to Frankie directly.

⁣Get BlueMail for Android ​

frankiearzu, 2023-03-17:
Lionel!
I think i got it working!!!
There was a big bug in some of the original code on how to process some BCD values.  That's why we were getting crazy values.

New firmware here:  (GPSFix4)
https://github.com/frankiearzu/DSMTools/tree/main/EdgeTx_Firmware

Sensors: GAlt, GPS (LAT/LOG), HDG, GSpd, Date (Only Time), Sats
Also have the Flight Mode comming from the SAFE/AS3X data.

The LAT/LON, need to validate the seconds. The RAW data comes: RAW data 28o 12' 7154", LON: -82 09' 8040"
Dont know what the secons 7154 means exactly.. Cannot be 71 sec, or all milliseconds.

[image]
[image]

lionelkeeping, 2023-03-19:
Frankie, I can confirm your new firmware accurately provides the precise GPS position, both in DD MM.mmmm format and in DD MM SS.ss format dependent on the radio setting of NMEA or DMS.  The GPS seems to be very accurate, showing a position within a few feet of the actual location, though altitude varies widely which seems to be normal for a GPS anyway.  See attached screenshots of my Zorro telemetry screen and of the data screen provided from the TELEMETRY page of the Zorro.  Thank you for your work on this!

[image]
[image]
[image]
[image]

Neutral67fr, 2023-03-20:
I can confirm the issue is solved, I can now get the correct data from the SPMA95871 Spektrum GPS. There is a little offset in altitude but it's something quite common between different GPS units due to lower accuracy for altitude.
Its not visible in the screenshot but I also get a correct Sat count (13 to 16 at the moment).
Thanks Francisco for your valuable work!
[image]

closed 2023-03-23 by pfeerick

TEH-ArcGIS, 2024-07-07:
Frankie, thanks to you and lionelkeeping!  I’ve bumped into a similar situation. I can capture GPS reading using a spektrum nx10 and a spektrum rx ar630.  When I bind to the RadioMaster, it doesn’t pickup any GPS values.  I have a skyid. I was on edgetx 2.10.2.   I went back to edgetx 2.9.4 to see if that made a difference.  It doesn’t…. Just no luck in getting gps values.  Just not having luck trouble shooting this issue.

frankiearzu, 2024-07-07:
With the SkyId, it sends GPS data on the regular GPS telemetry data channel, and the User/DRONE-ID on a separate message (message 27xx, you probably will see some sensors with this names 2700, etc).

Are you geting telemetry for other sensors?  the AR630 should give you gyro info as well as safe info (all the 05xx sensors).

TEH-ArcGIS, 2024-07-08:
Thanks Frankie for your time.

Yes, I am getting telemetry from other sensors.

Something is blocking GPS, Gpalt, Gpspeed, and satellites.  It worked when I first got the SkyId.

Just can’t account for the absence of the GPS data????

Thanks Frankie!!

Tomorrow Hall

qsiguy, 2024-07-29:
Been a couple months since I used my SkyID but I have encountered the same issue.  No longer getting GPS data from the SkyID.  I assumed it was due to the recent Spektrum 3.x firmware update but based on the post above from 3 weeks ago it was apparently an issue prior to that.

TEH-ArcGIS, 2024-07-29:
Haven't been able to put my finger on it or get a final answer.......

SkyID works on the Spektrum, but not on RadioMaster.....................

***@***.******@***.***>
515.664.7674
 Population is growing @ 1.1%/year
Food demand is growing @ 1.75%/year
Energy demand is growing @ 2.0%/year
Water demand is growing @ 2.7%/year

qsiguy, 2024-07-29:
Sounds like we've attempted about the same things to narrow it down.  My NX6 works with the SkyID as well.  I reverted back to 2.9.4 no luck, then reverted back to 2.9.3 also with no luck.  Only thing I cannot "undo" is the 1.1 firmware update to the SkyID.  I wish I could get it back to 1.0 to test that as that is the only variable I cannot retry.

On the DSM Smart RX Lua "RemoteID" do your values jump around?  Mine to now, they previously did not.

TEH-ArcGIS, 2024-07-29:
Yes, did the exact same thing!!!!

Missing something……….

frankiearzu, 2024-07-29:
Looking into it..   don't have SKYID module, but what we can tell from a capture done by @qsiguy, is that the RX is not sending the original GPS data..   I guess they move it to another message.    Probably lets open a new ticket to track this..     This ticket was closed long time ago.

frankiearzu, 2024-07-29:
Move the conversation to #5350 as a new bug