EdgeTX/edgetx #3301: strhelper function hex2char() returns wrong characters for 0xa..0xf

mha1, 2023-03-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

radio/src/strhelper.cpp:
`char hex2char(uint8_t hex) { return (hex >= 10 ? hex - 9 + 'A' : hex + '0'); }`

outputs for
```
  for(uint8_t hex = 0x00; hex < 0x10; hex++)
    printf("hex %x -> \"%c\"\n",hex, hex2char(hex));
```
```
hex 0 -> "0"
hex 1 -> "1"
hex 2 -> "2"
hex 3 -> "3"
hex 4 -> "4"
hex 5 -> "5"
hex 6 -> "6"
hex 7 -> "7"
hex 8 -> "8"
hex 9 -> "9"
hex a -> "B"
hex b -> "C"
hex c -> "D"
hex d -> "E"
hex e -> "F"
hex f -> "G"
```

### Expected Behavior

Well, you know ...

### Steps To Reproduce

checkout `hex - 9`

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

pfeerick, 2023-03-05:
Nice catch. Insidious little gremlin... looks like it was probably due to `hex2zchar` seemingly doing something similar, but must be handled differently as that is legacy OTX stuff.

closed 2023-03-05 by pfeerick

mha1, 2023-03-05:
@pfeerick speaking of zchar stuff, isn't all zchar code obsolete?

pfeerick, 2023-03-05:
Most, but I'm not sure it's *all* obsolete - there seem to be some active references to it in some odd places still... like `strcatFlightmodeName` 🤷

mha1, 2023-03-05:
I rephrase: shouldn't zchar code be obsolete? The fact zchar code is only rarely used made me suspicious. 

Unrelated: I need some advise on another thing I'm implementing and testing right now. Are you available for a discussion on Discord DM? I'll jot down the general idea and some questions. Talk tomorrow or so.

pfeerick, 2023-03-05:
It was a pet hate of Raphael's and he is responsible for so little of it existing now ;) 

Sure, the next couple of days should be fine. Or you can stop by the dev hour chat thingy that's happening shortly and see what the guys think ;)

mha1, 2023-03-05:
I sympathize.

I'd prefer to have your opinion and advise as a first step if that's not a problem for you.

PR #3302 (2023-03-05): fixes hex2char() returning wrong characters for 0xa..0xf
Fixes #3301

Summary of changes: 
subtract 10 instead of 9 in case hex value is in [0xa..0xf]