EdgeTX/edgetx #3303: Zorro- Display time on Main and Telemetry screen.

KC748, 2023-03-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

 Zorro's main screen does not display the Time.

Relate:d https://github.com/EdgeTX/edgetx/discussions/3190

### Describe the solution you'd like

I realize there isn't a lot of real estate on Zorro's screen, but if possible...

1. By default, display the Time on the main screen. Next to the Voltage display would be ideal. Smaller text size for the Time and Timers would be OK.

[image]

2. Also by default, display the Time on the Telemetry Display screen. Ideally, at the very top, before or after the Battery Voltage.

[image]

### Describe alternatives you've considered

1. On the main screen, have the option to select Time, the way Timers are selected. 
2. On the Telemetry screen, the Time is a selectable option, to be displayed, from the Display screen. This is OK as is.  

### Additional context

A quick glance at the clock on the main screen is a convenient way to squeeze in the last minute of flying before having to pack it up to stay on schedule and not upset the wife. :)

philmoz, 2023-03-04:
There's not really a lot of room to play with.
The flight mode and RSSI are displayed next to the battery, and a long model name will fill the top section.

[image]

philmoz, 2023-03-04:
The telemetry screen would display timer 0 to the right of the battery voltage if the code did not have a bug in it.

[image]

KC748, 2023-03-04:
Oh I see. The Main screen is jammed full. I guess I'll continue using the Telemetry screen for the Time.
Thanks, Phil.

philmoz, 2023-03-05:
I can squeeze the time in on the main screen with a bit of re-arrangment. It's crowded if you have long model names.
I had to move the 'unexpected shutdown' indicator (!) to the top left.
Also the left vertical trim box will slightly overlap the model name when set to max trim (which should be uncommon I would think).

[image]

philmoz, 2023-03-05:
I can move the battery to fit timer0 and current time on the telemetry screen.

[image]

pfeerick, 2023-03-05:
Funnily enough, I want a longer model name on B&W  ... `TinyTrainer` instead of `TinyTraine` ... which would make things even worse! 🤣 

I wonder if we can play with the timer display main view... it only shows two of the three configured times, so if you push the second timer over to the right so it's aligned with the first, could shove a clock in there? 

[image]

philmoz, 2023-03-05:
What about this - if we make the trim bars slightly shorter and put the time between them
This way the time is visible on all the screens (except channel monitor).

[image]

pfeerick, 2023-03-05:
That works :) Yes, it'll be smaller on the smaller LCDs like the T-Lite... but no smaller than the other text... reading the time is the least of the problems there! 😆

Eldenroot, 2023-03-05:
Cool

KC748, 2023-03-06:
Oh, wow! Thanks guys. That’s perfect.

PR #3306 (2023-03-10): feat(b&w) - Show time on the main screen and telemetry screen.
Fixes #3303 

See the above issue for discussion and screen shots.
closed 2023-03-10 by pfeerick