EdgeTX/edgetx #3309: Curve types not translated in Inputs or Mixes

ulfhedlund, 2023-03-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The curve types are hardcoded to "Diff", "Expo", "Func" and "Cstm" and can not be translated.
It's the same for B&W and color radios.

### Expected Behavior

The curve types are not hardcoded and can be translated.

### Steps To Reproduce

Go Model Settings - Mixes or Inputs
Edit any line
Go down to Curve and click in the drop down list to the right.

### Version

2.8.0

### Transmitter

FrSky X9 Lite / Lite S, Radiomaster TX16S / TX16SMK2

### Anything else?

@ParkerEde DE
@Eldenroot CZ
@robustini IT
@zyren CN / TW
@ajjjjjjjj PL
@ToshihiroMakuuchi JP
@HThuren DA

If you can provide translations for the following I'll include them in a PR.
#define TR_VCURVETYPE   "Diff","Expo","Func","Cstm"

HThuren, 2023-03-05:
DA
Those are very techincal, so
#define TR_VCURVETYPE "Diff","Expo","Funk","Cstm"

zyren, 2023-03-06:
******************************************************
CN

#define TR_VCURVETYPE "单边","指数","函数","自定义"
******************************************************
******************************************************
TW

#define TR_VCURVETYPE "單邊","指數","函數","自定義"
******************************************************

robustini, 2023-03-06:
IT

#define TR_VCURVETYPE "Diff","Espo","Funz","Pers"

ulfhedlund, 2023-03-06:
> DA Those are very techincal, so #define TR_VCURVETYPE "Diff","Expo","Funk","Cstm"

Yes, but they could nevertheless be translated. I have used "Egen" instead of "Cstm" in Swedish. That may work for Danish as well, perhaps.

ParkerEde, 2023-03-06:
DE
#define TR_VCURVETYPE "Diff","Expo","Funk","Ind."

Pat6874, 2023-03-06:
FR
#define TR_VCURVETYPE "Diff","Expo","Fonct","Perso."

ulfhedlund, 2023-03-06:
@Pat6874: could you have a look at PR #3281 and provide FR translation for that as well?

Eldenroot, 2023-03-06:
We cannot use more symbols? Quite difficult for CZ...

ulfhedlund, 2023-03-06:
For color radios you could, I suppose, but not for B&W radios with small screens.

HThuren, 2023-03-06:
you can use TR2(x, y), where y will be used on larger screens

Eldenroot, 2023-03-06:
@HThuren maybe we should add this to all languages? :)

ulfhedlund, 2023-03-06:
@HThuren is right and this is use in many places. But there isn't actually that much more space available for color radios in this case, so we'll keep it the same for all radios.

ulfhedlund, 2023-03-06:
Only PL language left now.
@ajjjjjjjj - do you have any translations to provide before I make a PR?

pfeerick, 2023-03-08:
@Pat6874 Would you be interested in doing other translations for FR? If so, I can add you to the list to be pinged when there is a new translation needed.

closed 2023-03-08 by pfeerick

Pat6874, 2023-03-08:
Ok, it's possible.
Patrick