EdgeTX/edgetx #3318: AFHDS3 iBus2

jmxp69, 2023-03-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

EL18. EdgeTX 2.8.1. Int Module. FTr16s. Module Option: External Bus Type iBus2 will not bind. iBus and SBus bind fine. 

### Expected Behavior

Should be able to switch bus option to iBus2 and get a bind as it does with iBus and Sbus.

### Steps To Reproduce

Module Option: Set external bus type to iBus2. Bind.

RX and TX bind quickly when using anything but iBus2.

### Version

2.8.0

### Transmitter

Flysky EL18

richardclli, 2023-03-07:
Flysky receiver FTr16s did not support ibus2, that is why the binding is not success.  However, there are some problem in keeping UI in sync with the settings stored in the receiver.  ibus2 should not show in the module options.

jmxp69, 2023-03-07:
From my contact @ FlySky: "All AFHDS3 enhenced edition receivers both support ibus2. But for EL18 radio, since it works with EdgeTX instead of our system, it doesn't support ibus2 signal output."

So yes, it appears ibus2 should be removed as an option.

pfeerick, 2023-03-08:
Hm... Flysky need to update their specs and manual sheets also, as there is no mention of IBUS2 for the FTr16s (or for the FTr10 for that matter)... how is an end user supposed to know these things? 😆 

[image]

pfeerick, 2024-03-05:
@richardclli Does `iBus2` still need to be removed?

richardclli, 2024-03-05:
> @richardclli Does `iBus2` still need to be removed?

Well, no ibus2 is possible in EdgeTX anyway.  Still waiting for Flysky firmware update.  they promise to open up ibus2 for opensource TX protocol.

richardclli, 2024-03-05:
> @richardclli Does `iBus2` still need to be removed?

And I remember you should not be able to select ibus2 since v2.9.0