EdgeTX/edgetx #3321: Lua function script run twice on X9D+ works correctly in Companion

stephendevelop, 2023-03-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I put a LUA script in SCRIPTS/FUNCTIONS folder
I make it run via "special functions" "Lua Script" action and put it behind a switch.

In the companion, the lua script is executed only once
On the taranis X9D+ running Edge TX  v2.8.0 this is run twice.  
On the Logical switches screen on the taranis, the Logical switch is only blinking on and off.

Tried to put it behind a logical switch or behind an edge switch, same behaviour, always run twice on the taranis X9D+, only runs once in the companion

### Expected Behavior

Should only run once on the TaranisX9D+ and the behavior should be consistent between taranis and companion

### Steps To Reproduce

1. put simple lua script ( playsound ) in functions folder
2. create logical switch that goes on for one second after a switch is down
3. create special function that is activated on logical switch, action is the Lua Script

### Version

2.8.0

### Transmitter

FrSky X9D+

philmoz, 2023-03-09:
Lua scripts are run every 50ms.
Your logical switch is on for 100ms (0.1s) so the script in the special function can run twice before the switch turns off.

On my Mac, the simulator will sometimes run the script twice.

The 'play script' special function should probably have the '1x' repeat functionality implemented to handle this case.

stephendevelop, 2023-03-09:
Thank you for the explanation!

Maybe there's a better way to implement what I want to achieve:

What I want to achieve, is a LuaScript that plays one track and then increment a counter, on every flip of the momentary switch.
So that every time I flip the momentary switch, the next track is played
I want to use this to play the F3A pattern figures one by one.

lshems, 2023-03-10:
You can perfectly make a function that only runs once bases on a variable déclarée outside thé function to check if it has run already. Search upvalues lus on Google.

PR #3338 (2023-04-10): feat(radio) - Add 'repeat' option to 'Lua Script' special function.
Fixes #3321 

Adds the 'repeat' option to the 'Lua Script' special function.
The options are:
- 'On' = current functionality where script runs repeatedly while SF switch is active.
- '1x' = script is run only once for each activation the SF switch

Companion, Color and B&W radios supported.
Some minor layout adjustment needed for 128x64 B&W radios.

[image]

[image]

[image]
closed 2023-04-10 by pfeerick