EdgeTX/edgetx #3324: 6P multi pos button can no longer play soundtrack

david9981, 2023-03-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After upgrading from v2.7 to v2.8.1 on my TX16sMKII, the 6 pos buttons can no longer play sound track using Special Functions.

### Expected Behavior

switches such as 6P1, 6P2, 6P3, 6P4, 6P5, 6P6 should be able to play track and function like any other switches.

### Steps To Reproduce

NA

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

david9981, 2023-03-08:
The audio files are fine..I can play those sound track using other switches, just not the 6P1 pos. switches. I tried using other audio files, still no luck with the 6pos buttons.

In fact, I can't seem to assign any functions to those six pos. buttons under Special Functions. It worked fine in v2.7.

philmoz, 2023-03-08:
Very strange.

Can you zip up the MODELS and RADIO folders from the SD card and upload them here.

david9981, 2023-03-08:
[tx16s.zip](https://github.com/EdgeTX/edgetx/files/10919279/tx16s.zip)

pls see attached.  I've tried to start fresh and create new model, new special functions. still unable to get the TX16s 6 pos. button switches to play track or function.

david9981, 2023-03-08:
Problem solved! Thanks for checking.  It turns out I need to calibrate the 6 pos. button switches on the TX16s, which is counterintuitive since there is no way to "center" 6 pos. buttons.  Besides, without calibration, the 6 pos. button still works and sending outputs (albeit from -100% to around -10%, instead of -100% to +100%). Nevertheless, the special functions shouldn't be dependent on switches' output but look solely at the pressing of the button itself.

philmoz, 2023-03-08:
You beat me to it - I was just going to suggest that as there was no calibration data in your radio.yml file for the 6POS switch.
You should calibrate the pots and sliders as well.

AFAIK the 6POS switch is not a set of buttons; but appears like a slider to the radio - hence the need for calibration.

gagarinlg, 2023-03-08:
The 6 Pos on the TX16S has an extra MCU with a closed source firmware by RM. All the main MCU gets is an analog voltage representing the currently active position.
Currently the 6 pos has to be calibrated, because it has handled as an analog input. In the future the need to calibrate it will be removed, because all TX16S have very similar values and we can hard code ranges.

closed 2023-03-09 by pfeerick