EdgeTX/edgetx #3334: Throttle Lock Function

jmxp69, 2023-03-09:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Throttle Lock Function - GF or SF? 

### Describe the solution you'd like

Having been on OTX/ETX for the last 8 years, I've been shielded from throttle lock features on other radios. Recently while exploring the PL18 and after talking to a friend on Futaba, I found a throttle lock function that I believe should be made available on EdgeTX. For the dev team that have PL18 firmware, you can easily examine how it works; however, I will provide an outline.

From Locked State:
SWA Up - Throttle is locked. FlySky firmware requires all switches to be up/away on boot or they trigger a warning.
J3 Down - Throttle stick is in down position.
SWA Down - Throttle is unlocked, and throttle channel is active.

SWA Up - Throttle is locked.
J3 !Down - Defineable range, but just not down.
SWA Down - Throttle remains locked.
J3 Down - Throttle unlocked, and channel is active.

After Unlocking:
SWA Down = Unlocked
J3 !Down - Defineable range, just not down.
SWA Up - Throttle remains active.
J3 Down - Throttle now locked. 

On SF the function added could be Throttle Lock.
Variables would be Throttle Channel, Throttle Input, and Throttle Lock Switch (should be any physical or logical switch).

I am acutely aware this can be programmed with logic--I've down half of it. The problem is, how many people will go through the trouble of creating this logic for every model and using it? If it were a function where the user needed to only identify their throttle stick, their throttle lock switch, and the throttle channel, it would probably increase the safety factor of EdgeTX radios substantially.

PS I'm told by my Futaba friend, this is exactly how it works on his Futaba radio as well.

### Describe alternatives you've considered

https://www.youtube.com/watch?v=buqQNZVoUXQ
https://www.youtube.com/watch?v=Y0eKFAFNeX8
https://www.youtube.com/watch?v=OjDsJRsk4ak

I am fully aware of the logic capabilities of the radio. I'm asking for throttle lock to be made a function to enhance adoption and improve safety.

richardclli, 2023-03-10:
Well, you can do whatever locking unlocking logic via powerful mixer settings.  I fly heli and I programmed a lock that can only unlock the throttle when throttle is down and idle switch in 0 position.

richardclli, 2023-03-10:
And if you want a simple lock/unlock, it can be done with only one line in the output, in which a switch can force the throttle output to -100.

jmxp69, 2023-03-10:
Richard, This is John from RC Video Reviews. I'm pretty familiar with Edge :)   Of course the logic exists to do these things--I've made videos showing how. As a matter of safety, I believe adoption would be substantially higher if made available as a built-in function as on PL18.  Yes, a single line Override is the most basic simple lock on Edge. Wouldn't a built-in proper locking function be better though?

elecpower, 2023-03-10:
A new SF/GF still requires the pilot to know it exists and implement. An alternative method, still requiring to know it exists, would be a lua script that prompts for the required variables and creates the LS rows and SF/GF rows.
There would be no data structure or code changes required. Also when the pilot becomes more familiar with EdgeTX they are then able to modify the lua created throttle lock.

jmxp69, 2023-03-10:
Access to the code for changes to suit would be nice; however, lua also usually requires an installation. Unless this lua script were considered foundational and rolled into the sd card contents so that it winds up on every deployment.

I prefer to avoid the mechanics of how because the dev team usually are pretty bright about these things. I'm far more interested in the what...

pfeerick, 2023-03-10:
I'm more inclined to go the way that Neil is suggesting, as the infrastructure is already there, it just needs to be made simpler to understand (rather than re-inventing the wheel). The thing to keep in mind though is that similar arguments can be made that dual rates, expo, etc, etc, are also "simpler" on other systems... although simplicity usually comes with limitations.

richardclli, 2023-03-10:
Well, I think the use of lock/unlock is not quite generic and different pilot may have some different settings.

I think Neil's suggestion is good, but why don't we think of something more interesting?

It would be nice if EdgeTX can support some kind of template setup that can be pre-build by experienced pilot and only expose a few parameters for newbie to choose (which switch control what, DR settings, etc.) and hides all the details behind.

pfeerick, 2023-03-10:
What, you mean like SoarETX? 🤭 Have a look at how Jesper set that up - and that's included as standard with the sd card packs for widescreen colorlcd radios.

HThuren, 2023-03-12:
Betaflight has a preset functionality, where almost all settings in betaflight can be envolved. The settings scripts has it own github project, since different programmers can develop. [https://github.com/betaflight/firmware-presets](url)
But there are a need for at command interface then in companion or radio, or both