EdgeTX/edgetx #3337: R9M ACCESS is not available on Jumper T16 after merge of #3055

MRC3742, 2023-03-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

"R9M ACCESS" is not available as a selection for the External RF on a External Access Modded Jumper T16.
Using self built firmware I have confirmed the problem to begin after the merge of #3055 new module/serial API.
Up to this commit the Module is working properly and is properly read on the Version page - Modules / RX version.

A few things I have noticed after problem occured with screenshots attached
(1) Radio Hardware Page - AUX1 Serial port - External Module is available here but "Port power" radio button is now missing.
(2) Model External RF Page - R9M ACCESS is not shown as a select-able option.
(3) External Module Power LED does not light if updating from an earlier ETX version with R9M ACCESS already selected.
(4) Changing internal Module from ISRM to Multi makes no difference.
(5) Changing back to the earlier commit and all is working again.
 
[image]

[image]

### Expected Behavior

This option to be available as I have already added external inverters to allow the Heartbeat and CPPM Module pins to work independently from ISRM Heartbeat trace path. This allows both Internal and External modules to operate simultaneously with each providing its own telemetry. (Bench tested only)

### Steps To Reproduce

Explained above

### Version

Other (Please specify below)

### Transmitter

Jumper T16

### Anything else?

R9M ACCESS is available using build from commit df00d7d8a02e57497a16ccc54e298f025fd41c8a
But stops working after build from commit 72193eda8260767b61c2945da353a503345bdaee

MRC3742, 2023-03-27:
Isn't this an issue with all X10 variants?
Can someone please test for this problem using an External ACCESS Hardware Modded  X10 or RM TX16S?

Thanks - Rich

pfeerick, 2023-03-27:
I think part of the issue is it seems to get filtered out of the menu options here even though AUX1 is configured "External Module"... 
https://github.com/EdgeTX/edgetx/blob/ebef94d993c59137f879751510056f9dfc6296ca/radio/src/gui/gui_common.cpp#L860-L863

@raphaelcoeffic Any ideas?

richardclli, 2023-04-02:
Can the AUX1 = EXT_MOD function can be generalized to support all protocols? i.e. Once selected the ports definitions will be different for external module e.g. To support the flysky protocol for FRM303 as well.

richardclli, 2023-04-02:
I think some code maybe missing to put the aux port to the ext module when the UI set this up.

raphaelcoeffic, 2023-04-02:
>(1) Radio Hardware Page - AUX1 Serial port - External Module is available here but "Port power" radio button is now missing.

This is because the port power is then controlled by the module implementation. So basically the model settings saying whether or not the module is enabled, and also when we need to switch the module ON/OFF for various reasons.

>(2) Model External RF Page - R9M ACCESS is NOT shown as a select-able option.

This is caused by the protocol selection being currently solely based on some static definitions in the code, and not yet using the module API to query available ports. Needs fixing, obviously. 

>(3) External Module Power LED does not light if updating from an earlier ETX version with R9M ACCESS already selected.

Seems we have a missing definition to actually enable the feature (it's called "configurable module port" internally; see `CONFIGURABLE_MODULE_PORT`).

>(4) Changing internal Module from ISRM to Multi makes no difference.

That is expected, these settings are not related in any way.

>(5) Changing back to the earlier commit and all is working again.

That is expected as well ;-)

richardclli, 2023-04-02:
See if I can work out a fix, for this and FRM303 as well.

PR #3442 (2023-04-15): Fix configurable port, solved R9M Access problem and support FRM303 via mod

Fixes #3337
Fixes #3413

Summary of changes:
1. Fix configurable port problems
2. Resolved R9M Access MOD unusable
3. Support FRM303 in TX16S via AUX1 mod (similar to R9M mod but no need inverters.

Documentation for mod: https://github.com/EdgeTX/edgetx/wiki/Flysky-FRM303-Mod-for-TX16S
closed 2023-04-15 by pfeerick