EdgeTX/edgetx #3340: ! Switches not showing in Widget Settings

jmxp69, 2023-03-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

EL18 - 2.8.1 release version.

Widget Settings -> Switch -> Scrollable list stops at Input Rudder. There are no inverted switch options like we see in all other switch selection dialogue boxes.

### Expected Behavior

Would like to see list of all ! switches - Including configured logical switches.

### Steps To Reproduce

Widget Settings
Switch Drop Down
Scroll Up

### Version

2.8.0

### Transmitter

Flysky NV14

### Anything else?

My monkey is laughing at me because I can't get a simple radio to do what I want...

He's relentless...
It's unkind...

pfeerick, 2023-10-16:
Hey @jmxp69, do you remember exactly *what* widget this was in relation to? It sounds like you were using a widget that was presenting a source list, not a switch list, but if that was the case, it should have shown all the stick inputs, switches, telemetry sensors, etc... just like a mixer source list.

jmxp69, 2023-10-16:
Sorry Peter, I don't remember. I should have documented it.

pfeerick, 2023-10-16:
No problem John. Lets call this fixed through #4190 then, which it should be once widgets start making use of the SWITCH choice now that that is fixed.

PR #4190 (2023-10-16): fix(color): 'SWITCH' widget option logic, enable for Lua widgets
Fixes #3340 

- Logic for SWITCH option was using unsigned values instead of signed values.
- Added missing setup code for Lua widget and constant for SWITCH option selection.
closed 2023-10-16 by pfeerick