EdgeTX/edgetx #3348: Misleading "RX channel order" on BW radio (possibly colorlcd too, I didn't check

3djc, 2023-03-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The working used is misleading and technicaly not accurate imho. All this field does is define a default RADIO channel order, but actual RX channel order might be affected by things like multimodule channel reordering, or receiver side pin mapping on ELRS and others. In no way can it be guarented that RX will actually use the order shown here.

[image]

### Describe the solution you'd like

Remove RX from the wording

pfeerick, 2023-03-14:
Color has the proper wording "Default channel order" .... has never been changed from the OTX wording, which I guess was done that way to struggle to condense "Default channel order" for 128x64?? 

Anyway, since 15 chars seems to to be the visual limit I'm tempted to change it "Def chnl order"

https://github.com/EdgeTX/edgetx/blob/b258c3b92c6b4bf0e6fb0d5801d6117fcdfa8519/radio/src/translations/en.h#L457

3djc, 2023-03-14:
Maybe 'Def chan order' ?

pfeerick, 2023-03-14:
Sounds good to me. I'll include that in #3326 as it will need some translations done anyway.

commit cf09c8b2 (2023-03-14): fix(b&w): "Rx channel ord" -> "Def chan order"
Resolves #3348
PR #3326 (2023-03-23): Page header unification
Resolves #3284
Resolves #3014
Resolves #3348
Resolves https://github.com/EdgeTX/edgetx/issues/3015

Summary of changes: 
 - Almost all mentioned pages have had their headers modified for consistency
- Exception: Hardware/calibration screen as I'm not adding a label to that since the entire calibration screen UI needs work
- Exception: Screens that have tab navigation, which should be brought up in a separate issue once there has been a proposal for changes to them.
- Also picks up "Edit Label" issue and changes to "Label model", which is consistent with other menu optio [... cut]
commit 335c0a3a (2023-03-23): chore(colorlcd): Page header unification and other cosmetics (#3326)

* chore: Change Radio setup page headers

* chore: Model Setup page header changes

* Main menu, Model Manager, Model Templates
Including needed translation strings

* Model telemetry sensor edit page
Taking into account sensor value shown in header

* chore: Change "Edit Labels"
Resolves #3014

* fix(b&w): "Rx channel ord" -> "Def chan order"
Resolves #3348

* chore: Update translations

* chore: Change to prefered "Manage Models"

* chore: Cleanup

* chore: Move some colorlcd-only strings
May or may not save some flash mem [... cut]
closed 2023-03-23 by pfeerick