EdgeTX/edgetx #3351: provide bind function on a switch in special functions

ror1948, 2023-03-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

some tiny quads require a bind every time you turn power on. The one I was this on is Proto X quad using husban protol. 

### Describe the solution you'd like

Be able to program a switch to put Tx into bind mode as long as switch is active but switch off the bind once it is bound or switch position is at do not bind.

elecpower, 2023-03-14:
There are a couple of what are categorised 'dangerous functions' and thus hidden for binding internal and external modules. Requires a custom build to expose. I am unsure whether the radio code to action still exists or is up to date across all radios.

One for the radio gurus.

pfeerick, 2023-03-14:
> I am unsure whether the radio code to action still exists

Yes, FUNC_RANGECHECK and FUNC_BIND still exist behind `DANGEROUS_MODULE_FUNCTIONS` flags, but I have no idea as to if they function or not still.

ror1948, 2023-03-14:
Yes it does sound dangerous and the quad is just a small toy.
Please feel free to close my request.
Thanks

- Ralph

pfeerick, 2023-03-14:
Can you give us an update later as to whether the MPM bind on channel functionality is still working properly ;) I've never actually used it myself - just knew it was there, and is supposed to disable the auto-bind on power on/model select bind behaviour, and tie it to a channel so could be controlled via a switch.

closed 2023-03-14 by pfeerick

ror1948, 2023-03-14:
Sure if you can tell me how
Thanks

- Ralph

richardclli, 2023-03-14:
This reveals what bind on channel works, ha.  Notice this function for some time already but never know how it works, now everything clears.

pfeerick, 2023-03-14:
I gave the link to the MPM documentation earlier ;) - https://www.multi-module.org/using-the-module/binding#bind-on-channel - it says to toggle the "Bind on channel" option for the model in question, and then assign a switch to CH16 (or a LS condition that toggles CH16).