EdgeTX/edgetx #3356: Zorro- While selecting Protocol Type, Unable to scroll in reverse from Config.

KC748, 2023-03-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

In Model Setup, it is possible to scroll past the last protocol, (XK currently) to the beginning of the list, Config. However, from Config, it is not possible to scroll in reverse back to the last protocol.

Over shooting the last protocol is quite easy and it's a little annoying having to scroll all the way back to the end of the list from the beginning. 

[image]

### Describe the solution you'd like

Allow scrolling in reverse from Config to the end of the protocol list. 

### Describe alternatives you've considered

1. Place a hard stop at the end of the protocol list with a haptic and/or beep to signal the end.

3djc, 2023-03-15:
I think that would require a multi module firmware change, as on BW screen, the module pilots the behaviour

KC748, 2023-03-15:
Ah, ok. Thanks.

closed 2023-03-22 by KC748