EdgeTX/edgetx #3362: Alignment of radio and model files between Companion and firmware

philmoz, 2023-03-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

I was comparing the YAML files generated on my TX16S to what is saved by Companion in the .ETX file and noticed a few differences.

Any differences can mean that restoring radio and model settings to a radio from Companion might mean data is missing or changed.

Before I plow ahead and change things I wanted to be sure I'm on the right track by getting feedback from people more experienced with this stuff.

### Expected Behavior

Restoring settings from Companion to a radio should result in equivalent settings be saved.

### Steps To Reproduce

The differences I found so far:

RADIO.YML:
- Companion is missing the 'manuallyEdited', 'alarmsFlash', 'imuMax' and 'imuOffset' properties.
- Firmware does not write empty (all 0) nodes to the YML file. Companion writes everything. Noticeable examples are calibration data where Companion will write empty calibration for every possible input the radio could have.
- Companion writes 'currModel' and 'contrast' and 'backlightColor' for all radios - should  only be for B&W radios.
- Companion writes 'gyroMax' and 'gyroOffset' for all radios; but these do not exist in the firmware.

MODEL.YML - This appears to be well aligned except for the module data which has some concerns:
- The firmware has a module data type of 'raw' which appears to be only used for conversion of some eprom saved data on some radios. It does not appear in Companion; but probably does not need to?
- PPM module type looks ok.
- MULTI module type is missing 'receiverTelemetryOff' and 'receiverHighChannels' in Companion.
- PXX module type is missing 'receiverTelemetryOff' and 'receiverHighChannels' in Companion (fields are present but not used).
- SBUS module type. Companion has no equivalent data structure; but uses the ppm.frameLength property for the sbus 'refreshRate' value. Companion is missing the 'nonInverted' property.
- PXX2 module type looks ok; but is called 'access' in Companion.
- FLYSKY module type is  missing in Companion.
- AFHDS3 module type exists in Companion; but does not match the firmware dats structure.
- GHOST, CRSF and DMSP module types all look ok.

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

elecpower, 2023-03-17:
... and this is why I rabbit on about PRs that only apply half a change and no one goes back to align Companion. This occurred in OTX too so some of the differences are carry over eg pxx2 vs access. Then we get to this point and questions arise as to why is there misalignment. Gripe ends here.

Some data struct fields are to handle importing bin files but ETX does not care anymore. Though Companion and ETX should be mirrors except for output sequence. However both yaml parsers do not care if they have extra data thus why often lack of radio checking before writing. Missing data is what catches us out.

elecpower, 2024-01-30:
> Companion is missing the 'manuallyEdited', 'alarmsFlash', 'imuMax' and 'imuOffset' properties.
Companion writes 'gyroMax' and 'gyroOffset' for all radios; but these do not exist in the firmware.

I believe firmware 'imuMax' and 'imuOffset' equates to Companion  'gyroMax' and 'gyroOffset'. Since Companion does not use the fields then rename fields and read/write in Companion as per firmware.

Thanks @philmoz for your detective work. I'll address all of these post adc refactor due to the conflicts that would emerge.

elecpower, 2024-02-06:
'alarmsFlash' is mapped to flashBeep in Companion

elecpower, 2024-02-17:
Modules are a mess including trainer config and using fields from other modules eg PXX.
I'll refactor to clean it up in preparation for a GUI rework I started ages ago.

closed 2026-03-28 by philmoz