EdgeTX/edgetx #3363: Companion re-orders my model list.

BarryCole, 2023-03-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

 I imported my models from OTX, OK BUT. I had a few models down in the 50-60 area on OTX, when I import them into Edge they all go to the top numbers.

On the Tx, I can move a model from say position17 to P60, but when I open Companion and upload from the Tx, the model moves back from P60 to P17.

Tx is FrSky X9D+   ETX is 2.8.1  Companion is 2.8.0

### Expected Behavior

I expect the model list to be the same in Companion as it is on the Tx.

### Steps To Reproduce

See Error

### Version

Other (Please specify below)

### Transmitter

FrSky X9D+

### Anything else?

ETX 2.8.1

pfeerick, 2023-03-17:
Someone more experienced with Companion needs to check this, but I think this is because companion re-numbers all the models when written, thus if model16 is the last model before model60, it gets moved to the next available slot. This isn't an issue for colorlcd radios, as they have an index file to determine what order they should be in, etc, but I'm pretty sure B&W radios rely on the filename to work out which line to list the model.

elecpower, 2023-03-18:
Confirmed Companion write models and settings to radio creates file names aligning with slot minus one (zero based). 
Confirmed the issue arises on Companion read models and settings from radio. Ignores perfectly valid file names as basis for model slot and sequentially orders them.
Confirmed launching the radio sim from Companion the models maintain their correct slot positions.

elecpower, 2023-03-19:
Further testing reveals same issue occurs opening an etx file

PR #3373 (2023-03-27): fix(cpn) - load models into slot based on filename for B&W radios
Fixes #3363
closed 2023-03-27 by pfeerick