EdgeTX/edgetx #3364: Use internal and external HF module at the same time

ParkerEde, 2023-03-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Horus X10S Express:
If I want to use the internal ISRM and an external MPM IRX4+ at the same time, this is not possible. As soon as the external one is set up and ready for use and then the internal ISRM is activated, the connection to the external MPM is immediately interrupted.

### Expected Behavior

both HF moduls should work at the same time

### Steps To Reproduce

try to use both internal and external HF Moduls

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

pfeerick, 2023-03-17:
Have you actually ever been able to do this on the X10? 

i.e. Due to architectural design choices in OTX which ETX has inherited (but there are efforts to change this entirely) not having treating the two (or more) RF modules completely separately, on the TX16 you can use internal MPM and external MPM together, but you can't use internal MPM and external CRSF (or vice versa). Well, actually you *can*, but only with very specific settings. And the same goes for some other module combinations.

MRC3742, 2023-03-17:
It can be done but requires a mod to disconnect the External Sport line from the internal ISRM.
Either by cutting the trace to ISRM or removing the pin on the internal ISRM.
It's believed to have been done intentionally to block just what your trying to do.

More info in these posts
https://www.rcgroups.com/forums/showpost.php?p=44546553&postcount=17686
https://github.com/opentx/opentx/issues/7181#issuecomment-564418313

Rich

ParkerEde, 2023-03-17:
Ah okay. 
If you remove the Sport pin, can you still flash the internal ISRM module?

MRC3742, 2023-03-18:
Yes - Tested working
Flashing the internal ISRM is unaffected by this mod.

Rich

ParkerEde, 2023-03-19:
@pfeerick should we close here? Or do you think we can do something about it software-wise here?

pfeerick, 2023-03-20:
Yes, I think it is safe to close it. This isn't the first time I've heard of the hardware mod needed for ISRM and SPort line (bypassing frsky's decision to disable *some*  external modules when the internal module is enabled), so there isn't a simple software workaround unfortunately - hardware modification is the only solution.

closed 2023-03-20 by pfeerick

anshchawla521, 2023-06-24:
Hi I have a RadioKing t18 (basically a jumper t18)(I know its not supported officially) and a Radio Master Tx12 (MK2 ELRS version) and on both when I enable the internal module the signal to external module (have tested with PPM and SBUS) gets changed and doesn't follow the rules of the protocol. Why might that be happening as now there is nothing related to Frsky ? and what would be the solution for getting a clean PPM / SBUS signal for the JR bay module.

pfeerick, 2023-06-25:
Because running two interfaces wasn't something really thought of when some of the underlying architecture was designed, weird stuff happens when say MPM and ELRS are both enabled, as they run at different speeds. This PR aims to try and resolve some of that https://github.com/EdgeTX/edgetx/pull/3642. I'm not sure what effect there is on PPM/SBUS.

anshchawla521, 2023-06-25:
Ohh ohkka. Thanks a lot , will look into the mentioned PR

ParkerEde, 2023-12-18:
> It can be done but requires a mod to disconnect the External Sport line from the internal ISRM.
> Either by cutting the trace to ISRM or removing the pin on the internal ISRM.
> It's believed to have been done intentionally to block just what your trying to do.

@anshchawla521 Do you have the possibility to plug an MPM into the ext. module bay of your moded X10ISRM and measure the S.PORT voltage for HIGH level with an oscilloscope or logic analyzer (analog input)?

anshchawla521, 2024-02-09:
@ParkerEde Sry for the late reply but I own a T18 which is very similar to a Jumper T18 , so I dont have the X10ISRM 😓