EdgeTX/edgetx #3367: Mixes being forgotten

sixtyfive, 2023-03-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When setting up a mix other than the 5 defaults (them being the 4 stick movements as well as Arm/Disarm on switch SC), and then rebooting the transmitter, any additions made to the Mixes tab are then gone again.

_Changes_ to one of the 5 defaults are retained, only _additions_ are forgotten.

### Expected Behavior

Adding a mix and rebooting retains that mix throughout reboots.

### Steps To Reproduce

1. Turn transmitter on
2. Long-press "Enter" button
3. Right-press right thumbstick to page 5/11 ("Mixes")
4. Down-press right thumbstick to CH6
5. Press "Enter" button
6. Down-press right thumbstick once
7. Press "Enter" button (selection blinks)
8. Operate SD button
9. Press "Enter" button (selection stops blinking)
10. Press "Exit" button, observe new mix is now set up
11. Press "Exit" button again
12. Turn transmitter off
13. Turn transmitter on again,
14. Navigate to "Mixes" again, observe new mix is missing

### Version

2.8.0

### Transmitter

BetaFPV LiteRadio3 Pro

### Anything else?

EdgeTX version is actually 2.8.1, but only 2.8.0 is selectable in the New Bug form.

VitalyKondratiev, 2023-04-11:
I didnt catch this bug on 2.8.1, but yesterday I upgrade EdgeTX to 2.8.2, and get this problem.
Downgrade to 2.8.1 saves the bug, but downgraded to 2.8.0 removes the bug
I have BetaFPV LiteRadio 3 Pro too.

oloendithas, 2023-04-22:
Experiencing the same issue on 2.8.2.

oloendithas, 2023-05-18:
Issue persists on 2.8.4

oloendithas, 2023-06-30:
2.10-nightly still not fixed

pfeerick, 2023-07-01:
I saw recent reference to this suggesting it only happens on the existing models, not new models... can you try creating a new model and seeing if this still happens. 

BTW, main / nighties are in no way safe to fly atm. Stick with 2.8.x or the 2.9 branch for now, other than for testing.

oloendithas, 2023-07-01:
Oh, indeed. Newly created model does not suffer from this issue (on 2.8.4).
Thanks for the clue!

paulo-serrao, 2023-09-03:
I am encountering the same issue, my radio is the RadioMaster Pocket.

Here is an example of a model with this issue, if I set mixer CH7, I loose it after a reboot (or simply switching to a different model). 

[model02.zip](https://github.com/EdgeTX/edgetx/files/12506937/model02.zip)

raphaelcoeffic, 2023-09-03:
@paulo-serrao thx for the file!

What happens here is the following:
- the file has been copied from another radio,
- which had *some* source that the destination radio does not have.

When such file is loaded in the destination radio, the following happens:
- the source is not found by the parser and set to `NONE` (`= 0`)
- when displaying the mixer lines, the first line with `Source = NONE` stops the display

If I make a quick & dirty patch to take the `Weight` into account as well, you can see your missing mixers:

[image]

Now the issue is that due to the how we store mixer lines in memory, it gets quite hard to patch so that it always works.
Historically, such mixer lines with `Source = 0` never happened, as you cannot choose it on the UI, and OpenTx would not let you load a file from another radio, as the sources were all just numbers and have a specific meaning only in the context of a specific radio.

raphaelcoeffic, 2023-09-03:
Oh, and btw: I believe this bug is present for quite some time now (probably as old as the YAML parser).

paulo-serrao, 2023-09-03:
@raphaelcoeffic, this is the original model file from X9D before I started making changes in the new radio to adjust to new button layout: [model02_x9d.zip](https://github.com/EdgeTX/edgetx/files/12508087/model02_x9d.zip)

The previous file I uploaded was after many changes on other menus/settings and also trying multiples times to set a mix on CH7 to Switch D.

PS: On X9D I had CH6 set to Switch G, that switch does not exist on RadioMaster Pocket, maybe that's the root cause? This is how it looks like on my Taranis

[image]
[image]

raphaelcoeffic, 2023-09-04:
> On X9D I had CH6 set to Switch G, that switch does not exist on RadioMaster Pocket, maybe that's the root cause?

This is it, yes. That transforms it into `NONE` (`---` on screen), which makes the UI think the mixer does not exist, and stops displaying any line after that.

oloendithas, 2023-09-11:
As was mentioned before, the workaround it to create a new model. The issue persists only in models created on pre-2.8 versions.

raphaelcoeffic, 2023-09-11:
@ValorLin @oloendithas as you can see, there is a code change proposed in #4008 which you can try (backup your SD card before, you will need it to go back to 2.9). We need feedback from you guys on that.

oloendithas, 2023-09-11:
> We need feedback from you guys on that.

I've recreated my model and can't reproduce the issue any more 🤷🏻‍♂️

paulo-serrao, 2023-09-11:
> I am using the BetaFPV LiteRadio3 Pro, After I upgraded to 2.9 yesterday, the issue occurs. Is there a workaround?

You can edit the yaml, clear out the section for the mixer and then recreate it on the radio. (this will avoid having to recreate the whole model and all settings)

raphaelcoeffic, 2023-09-11:
Exactly, the issue is the mixer source `NONE`. If clear that line, or set it to something meaningful, it works.

ValorLin, 2023-09-19:
> > I am using the BetaFPV LiteRadio3 Pro, After I upgraded to 2.9 yesterday, the issue occurs. Is there a workaround?
> 
> You can edit the yaml, clear out the section for the mixer and then recreate it on the radio. (this will avoid having to recreate the whole model and all settings)

Thanks! It works!. 
I believe that you guys have caught the problem.  @raphaelcoeffic 

Here is the original yml that caused this issue.
[model03.yml.txt](https://github.com/EdgeTX/edgetx/files/12664078/model03.yml.txt)

commit dd6c8c1f (2023-09-04): fix: mixer lines with empty source

Fixes #3367
commit 46a036e5 (2023-09-05): fix: mixer lines with empty source

Fixes #3367
PR #3999 (2023-09-24): fix: better iteration through mixers on display
Fixes #3367.
commit 47ff7284 (2023-09-24): fix(radio): Better iteration through mixers on display, sanitize invalid sources (#3999)

* fix: mixer lines with empty source

Fixes #3367

* Fix color mixes display.

---------
closed 2023-09-24 by pfeerick