EdgeTX/edgetx #3372: To much log files if using logical switches

FriedelXT, 2023-03-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

i try to switch on logging by using logical switch
(if rx volt > 2 then switch on L03. if L03 is ON, then start logging)
but doesnt matter what you use as trigger, the result is the same:

1 big log file plus every 5 seconds(?) a log file too.

If you switch on with hardware switch its working fine

### Expected Behavior

only 1 log file for each flight (switched logical switch on)

### Steps To Reproduce

[image]

[image]

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

FriedelXT, 2023-03-19:
[image]

mha1, 2023-03-26:
I tried this with 2.9 nightly. With RxBt at 4.9V a log is started, with RxBt falling below 2.0V (or telemetry lost) the log finishes. If RxBt comes back at a voltage greater than 2.0V a new log files is opened. This is as expected.

Can you please attach one of the short log files.

FriedelXT, 2023-03-26:
I did you a zip of all coherent logfiles. I switch on the tx and connect battery and then the drama began:
[nach dem einschalten.zip](https://github.com/EdgeTX/edgetx/files/11071982/nach.dem.einschalten.zip)
for workaround i switch on with 3step arm switch. off-log on -motor start. but more comfortable would be with a better trigger like connect with battery. have fun.

mha1, 2023-03-26:
At least 2.8 simulator shows the same correct behavior as 2.9 above. While logging please switch to the special functions page and observe the SFx indicator. Is it indicated as steady active or does it flicker? 
[image]

FriedelXT, 2023-03-26:
there is no flicker. but I dont know this simulation is real, because battery V not change

mha1, 2023-03-26:
Which module (internal MPM?) and protocol?

FriedelXT, 2023-03-26:
Internal MPM with frsky EU

mha1, 2023-03-26:
This happens while Voltage is sort of constant, at least above 2.0V all the time?

FriedelXT, 2023-03-26:
of course. with RSSI is the same problem. first i used RSSI >0

mha1, 2023-03-26:
can you please send your model.yml file

FriedelXT, 2023-03-26:
[Trex 550x.yml.zip](https://github.com/EdgeTX/edgetx/files/11072162/Trex.550x.yml.zip)

mha1, 2023-03-26:
L03 has a 5 seconds delay. Please remove the delay and try again.

mha1, 2023-03-26:
Found your problem:
- SF26 resets Telemetry based on L02 condition Cels > 24.8 (why do you do that?). As L02 has a set duration of 5 seconds you effectively reset the telemetry every 5 seconds as long as your flight battery is > 24.8 Volt. 
- Resetting the telemetry clears the RxBt telemetry value for a moment which leads to L03 being switched to off. After a short moment telemetry is up and running again. With RxBt now > 2V again, L03 goes to on after 5 second which starts a new log. 

This goes on and on as long as your flight battery is > 24.8V and will produce a new log file every 5 seconds as long as the flight battery is above 24.8 Volts.

Remove SF26 and it'll work as you expect it. 

Please close this issue as there is no problem with EdgeTX

FriedelXT, 2023-03-26:
Thanks for checking.
L02 means if Battery is full charged and i connect this for longer then 5 seconds, then reset timer and telemetry. That is in that moment when I  connect battery. Why every 5 seconds? 
This is not to forget reset before flight.
Then L03 switched on if more then 2 Volts (= ever)
So if I connect new battery, logs starts and telemetry do a reset.

FriedelXT, 2023-03-26:
> L03 has a 5 seconds delay. Please remove the delay and try again.

i delete this  delay and it looks that it works (on workbench).
I have only 1 file with 3 mb after longer time.

but this is crazy. i thought this delay means: if trigger is longer then......then switch is on

FriedelXT, 2023-03-26:
i did again a test...it works now how I expect. Thanks for helping

mha1, 2023-03-26:
please close this issue

closed 2023-03-26 by FriedelXT