EdgeTX/edgetx #3378: getRotEncSpeed() doesn't work

JimB40, 2023-03-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In 2.8.1 and nightly  getRotEncSpeed() returns always 1
In 2.7.1 returns 1 or 5 or 50 depending on rotatry encoder speed

### Expected Behavior

getRotEncSpeed() returns 1 or 5 or 50 depending on rotatry encoder speed

### Steps To Reproduce

1. run below script

```
local run = function(evt,ts)
  lcd.clear()
  lcd.drawText(0,0,getRotEncSpeed(),INVERS)

  return 0
end

return {run = run}
```

2. Rotate encoder

### Version

2.8.0

### Transmitter

Radiomaster TX16S / TX16SMK2

philmoz, 2023-03-20:
According to the commit history this was changed in commit c5160d8 on Jun 11 2022.
It is now only supported on B&W radios.

philmoz, 2023-04-09:
PR #3441 will (hopefully) address this.

The value returned for color radios will be a number between 1 and 100 depending on how fast the rotary encoder is turned.
I've never seen it go much over 50; but 100 is the upper limit currently set.

pfeerick, 2023-04-16:
It indeed does. I can get it to go all the way up to 100, and anywhere in between depending on speed.

closed 2023-04-16 by pfeerick