EdgeTX/edgetx #3380: TBS Agent Lite is slow to populate the Rx output map etc

ror1948, 2023-03-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

when running the TBS Agent lite 0.96 Lua script it is very slow when talking to the CRSF nano pro Rx with v6.19
Seems much faster when getting Tx info
Currently takes 40sec to display Rx output map

### Expected Behavior

display info of Rx within a few seconds
Currently takes 40sec to display Rx output map

### Steps To Reproduce

Tx(micro Tx module V2 installed in bay) is bound to a crossfire Rx at 400k
press SYS button
select TBS AGENT LITE lua script
Select Rx
Select output map

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

version 2.8.1

RipperDrone, 2023-05-06:
same here, X9D+. Script ran fine on OpenTX, completely freezes in NAno RX Output Map and Channel Map pages (they do not populate, even after 1min) :-(. Something is flawed in the memory management, it seems.

pfeerick, 2023-05-09:
Check the version number, this is from before the re-write which is in 2.9 ;) So not entirely sure what's going on there.

RipperDrone, 2023-05-13:
> Check the version number, this is from before the re-write which is in 2.9 ;) So not entirely sure what's going on there.

sry, this hint is still a bit cryptic to me - does it mean we should re-test once edgeTX 2.9 is out? Or is there an auto-nightly build branch where we can download a nightly version now already? Not versed to build my own 2.9'ish versions yet... :-)

pfeerick, 2023-05-14:
No, it is in response to the prior message, which is from JimB40 - he's talking about something that changed in 2.9 whereas the OP is talking about 2.8.1 ;)

And yes, you can test with the automated nightly builds if you wish - backup your MODELS and RADIO folders first just in case, and then flash one of the builds from https://github.com/EdgeTX/edgetx/releases/tag/nightly or use [Buddy](https://buddy.edgetx.org/#/flash?version=nightly&filters=includePrereleases) and enable prereleases under the filters.

ror1948, 2023-06-17:
TX16s
Still having a problem with this unable to populate output map because I can only set the rate to 115k..It will not bind at 400k.
This Rx did work before all this problem came to lite on CF 6.07 but now running 6.19 and Edgetx 
edgetx-firmware-nightly-d4b7f55e
thanks

pfeerick, 2023-06-23:
Closing this issue as it was brought up and appears to be resolved in #3688

closed 2023-06-23 by pfeerick