EdgeTX/edgetx #3381: TX16S IMU Cmake -DIMU=YES -DIMU_LSM6DS33 crashing USB connection indescriminately

Kevltan, 2023-03-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

 Cmake -DIMU=YES -DIMU_LSM6DS33 crashing USB connection indescriminately.

### Expected Behavior

To be able to use IMU to control screen wake up. This works fine but when used with USB cable plugged in the radio shuts down to Emergency mode indescriminately usually half way through either loading or uploading files to the radio's uSd card. Sometimes will shut down as soon as the cable is plugged into the PC! Have removed the Cmake addition and the problem goes away.

### Steps To Reproduce

Plugging USB cable into radio and PC whilst powered up using the  Cmake -DIMU=YES -DIMU_LSM6DS33 addition.

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

2.9 latest commit.

Kevltan, 2023-04-07:
Any update or solution to this problem please?

gagarinlg, 2025-01-24:
Is this still an issue?

Kevltan, 2025-01-24:
Yes I have removed hardware and cmake from radio now, as it makes it
unreliable.

3djc, 2025-01-24:
Does it require actual LSM6DS33 hardware installed or can you reproduce without ?

Kevltan, 2025-01-24:
Yes, I disconnected the hardware first and tried without and it still did
it. So removed cmake from the firmware build. It then worked fine when
connecting USB cable to PC. Before this when EdgeTX loaded it would
sometimes instantly crash with lead inserted and connected to PC. It was
almost a timing issue if the USB lead wasn't inserted and EdgeTx fully
booted to insert USB cable at the correct time without it crashing again.
It just became so  unpredictable I decided to remove both Hardware and
Cmake from build for safety reasons.

3djc, 2025-01-24:
Thanks, this should allow us to work on it more easily !

3djc, 2025-01-24:
Compiled current main with: -DPCB=X10 -DPCBREV=TX16S -DDEBUG=NO -DIMU=YES -DIMU_LSM6DS33=YES

Flashed radio and bootloader. Turn radio on, connected USB, copied the full English sound pack (that's like 6 minutes of SD copy, unplugged USB, all seems perfectly fine, I'm sorry to say that I could not reproduce the issue.

Kevltan, 2025-01-24:
OK all I can say is that since I have reverted all USB problems have
desisted!! I do not have answers the only thing I can say it is very
inconsistent and doesn't happen all the time but when it does it is an
absolute pain. Since I removed the IMU cmake the problem went away.

closed 2025-03-23 by Kevltan