EdgeTX/edgetx #3391: EL18 internal RF interface display suggestions

lolfling, 2023-03-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The interface display is disordered
[image]

### Describe the solution you'd like

Keep options organized

### Describe alternatives you've considered

Model setup-> internal RF \External RF\Trainer\Trmer1\Timer2\Timer3...

### Additional context

No response

pfeerick, 2023-03-23:
Actually, now that I think of it, for that screen orientation, having one button per row probably wouldn't hurt at all...

pfeerick, 2023-03-23:
Actually, no, this is outdated sorry - the layout issue has been improved in 2.9 by changing it to three columns for portrait screen.

closed 2023-03-23 by pfeerick