EdgeTX/edgetx #3403: Module ISRM-S-X10S Antenna selection options internal/external on model level effect global change

elekSENDER, 2023-03-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In a X10S with an ISRM upgrade modul at Model Setup/Internal RF/Module Options there is a possibilty in each model to change ext/int antenna. By changing this parameter from int to ext and viceversa this will effect not only the selected model but all models.

### Expected Behavior

The setted value should remain set and used for each model.
The other possibility is to move this kind of setting to a system setting page and remove it from model level.

### Steps To Reproduce

At model A select ISRM internal antenna.
Change at model B from internal to external antenna.
Check at model A antenna setting: it changed to external.

### Version

Other (Please specify below)

### Transmitter

FrSky X10 / X10S (ACCST)

### Anything else?

Firmware: x10-2.8.1-d9d4a8a
Transmitter: Frsky Horus X10s + ISRM upgrade module

pfeerick, 2023-03-25:
Just to be clear - does this mean you select Internal RF -> XJT or ISRM? As this changes how this option is managed.

elekSENDER, 2023-03-25:
Internal RF is ISRM.
And sorry, the above description is not completely correct: I can´t switch between int./ext. antenna, I mean the window where I can switch Ext. antenna on/off.

pfeerick, 2023-03-25:
I think the setting you are changing is a module-level setting, not a model or system setting. It also may in fact not change every model, but every model with the same receiver number. If this is a module-level setting, I don't think there is anything we can do about this. 

i.e. Is this an option shown when you go into here? (I can't see that in simulator since everything shown on that screen is determined by the module,  and I haven't used any ISRM/ACCESS stuff with dual antenna options.

[image]

elekSENDER, 2023-03-26:
> but every model with the same receiver number.

This occurs at 2 different models with 2 different RX with 2 different RX numbers.

Let me try to explain via screenshots direct from the transmitter:
Model A:
[image]

select button you signed above with red arrow:
[image]
ext. antenna is ON.

change to model B, different RX number and RX:
[image]

enter Module options:
[image]

switch Ext. antenna to OFF and save
[image]

Change back to model A:
[image]

Verify antenna settings:
[image]
previously exited with ON, now OFF.

elekSENDER, 2023-03-26:
I understand the changing on module level, but if there is no way to initalize the module differently with each model then it would be better to make this setting maybe anywhere around the module selection in the hardware section:
[image]

Just to prevent that you think you make the antenna selection by changing a model and screw off the external antenna without really changing the module setting.

pfeerick, 2023-03-26:
Yes, indeed - I'll find out if there is anything there that would be model level, and if not, we can then update this to be a "move the module settings to the hardware page" request.