EdgeTX/edgetx #3409: first rotary input on the encoder wheel is not evaluated

ParkerEde, 2023-03-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After the radio start, the first rotary input on the encoder wheel is not evaluated. This happens in both directions. Only the second input is then processed.
This is the case on a Horus X10, X12, X10Express and Taranis QX7 ACCESS.

### Expected Behavior

The first input on the encoder wheel should also be used immediately after a radio start.

### Steps To Reproduce

1. Start Radio
2. press ENTER, ENTER to enter the Modellmanager
3. scroll only one step to the neighboring model
4. observ, the fokus doesn't move. Only the second and further input is used

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky Q X7 / Q X7S (ACCESS), FrSky X10 / X10S (ACCST), FrSky X10 Express / X10S Express (ACCESS), FrSky X12

ParkerEde, 2023-03-27:
A better way to reproduce it safely is,
1. switch on radio
2. SYS button
3. turn encoder one click (no reaction follows)

pfeerick, 2023-03-28:
Also RM TX16S, so the rotary encoder driver generally.

mha1, 2023-03-28:
... which is not the best documented and intuitive part of EdgeTX (EXTI and Timer interrupts? buggy wiring?).

pfeerick, 2023-03-28:
And that's different to any other part how exactly? 😆 You won't need any documentation once you're been here long enough! 😜 

Buggy wiring is just that - IIRC it was the T16 and some other radios that have bad circuit layout, but a software fix was found to work around that. I'll throw you two bones though 😆
- I just checked the Zorro, since it isn't in that category - and it has the same bug, so either both sides have the same fault, or it's further back ;) 
- 2.8.1 doesn't seem to have this problem on either TX16S or Zorro (i.e. buggy and unbuggy), so you're looking for something that changed in 2.9

mha1, 2023-03-28:
Yeah, I know. Source code is the ultimate documentation.

[image]

pfeerick, 2023-03-28:
Well, that, and phone a friend when you get stuck 🤣

richardclli, 2023-03-29:
Does this relates to the buggy hardware?  As I know the driver is tweaked so that the buggy roller in TX16S works properly, and it works for fixed roller hardware as well.  Maybe @rotorman can answer more.

richardclli, 2023-03-29:
> 

2.8.1 no problem?  Only in 2.9?

pfeerick, 2023-03-29:
Please read https://github.com/EdgeTX/edgetx/issues/3409#issuecomment-1486531406 !!! ;)

richardclli, 2023-03-29:
> Please read [#3409 (comment)](https://github.com/EdgeTX/edgetx/issues/3409#issuecomment-1486531406) !!! ;)

I read the email push and reply too fast haha.

philmoz, 2023-04-07:
Some investigation:
- On my QX7s the problem is present in 2.8.1 so it does not appear to have been introduced in 2.9
- On my TX16S if I navigate to the Debug/Keys view without touching any buttons or the encoder, then turn the encoder slowly I can see it jump by one in the wrong direction, then back to 0. After that it works correctly.

philmoz, 2023-04-07:
I've found the problem in the rotary encoder on the QX7s (still checking if the TX16S is the same; but I suspect it is).

The initial state value read from the encoder on first rotation can vary on power up - it is not guaranteed to be a fixed value.
Therefore the rotary encoder driver needs at least two values initially to correctly determine the initial rotation direction.

In the current code using the first value read can select the wrong initial rotation direction. The second value then reverses this so the encoder looks like it has not moved.

philmoz, 2023-04-07:
I've posted a PR for this that seems to work well on my QX7S and TX16 Mk2.

However I can't trigger all the code paths for the ROTARY_ENCODER_SUPPORT_BUGGY_WIRING variation in the rotaryEncoderCheck function.

If anyone has a radio affected by this 'buggy wiring' issue can you please test the PR.

pfeerick, 2023-04-07:
`ROTARY_ENCODER_SUPPORT_BUGGY_WIRING` applies to TX12 (not TX12MK2) and the T16 family of radios- so T16 and TX16S but not T18 ;) So your TX16MK2 should have been using the `ROTARY_ENCODER_SUPPORT_BUGGY_WIRING` path, and the QX7S the normal one. Either way, I'll end up trying it in the next few days with the X9D+2019, TX16SMK1, TX12 and T-Pro for comparision. Although given its grey code encoders... it's not surprising that the state of both switches are needed before determining direction, which is why it is amazing this bug has been unnoticed for so long. Maybe just not many times the first click of the encoder is on a field where you notice it goes the wrong way (in the odd instance that it does).

philmoz, 2023-04-07:
> So your TX16MK2 should have been using the `ROTARY_ENCODER_SUPPORT_BUGGY_WIRING` path, and the QX7S the normal one.

Yes the TX16S uses the buggy wiring code block; but in all my testing I could not trigger the code in the extra 'if' check that then increments by INC_ROT_2.

gagarinlg, 2023-04-07:
They fixed the PCB with the last TX16S MKI batches and the MKII, so the faulty signals might not come in your MKII.

pfeerick, 2023-04-07:
If I'm tracing that code back properly that alternate path stems from #261. Does that seem right @rotorman? i.e. the INC_ROT_2 side is only relevant for transmitters with the buggy wiring, with the Alps encoder mod board?

rotorman, 2023-04-07:
The code support for correctly wired up encoder (with A and B quadrature encoder inputs wired the way they are supposed to) was added with #261, as the previous code only supported the buggy wiring with it's workaround.

PR #3445 (2023-04-16): fix(radio) - Fix direction detection for rotary encoder on startup.
Fixes #3409 

Delay initial direction detection until the 2nd input from the rotary encoder.

Tested on QX7S and TX16 Mk2.
PR #5194 (2024-06-21): fix(radio): various issues with rotary encoder.
Re-write rotary encoder logic and update Lvgl driver to fix:
- detection of initial direction of encoder after startup
- change of direction issue on radios with incorrectly wired encoder
- handle all wiring variations (without needing special #defines)

May fix #3409 

Candidate for 2.10.2
closed 2023-04-16 by pfeerick