EdgeTX/edgetx #3424: Transmitter blocked when external Antenna selected and no Antenna installed

icebreaker-ch, 2023-04-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have an Horus X12S. When I select "Hardware" -> "Antenna" -> "Internal and External" and no external Anenna is installed,
an Error Popup Box appears. The only way to get rid of it, is to hard switch off the transmitter by a long press on the power button.

### Expected Behavior

When the Error Popup Box appears, I expected, to press "RTN" and be back in the "Hardware" screen, to change my Antenna settings. 

### Steps To Reproduce

1. Make sure, no external Antenna is installed
2. Go to "Hardware" Settings
3. Set the "Antenna" settings to "Internal and External"
4. Try to get rid of the Error Popup Box
5. Long press Power button to hard switch off the transmitter

### Version

2.8.2

### Transmitter

FrSky X12

pfeerick, 2023-04-02:
Can you take a picture of this "Error Popup Box", or at least list the _exact_ wording used in the dialog?

icebreaker-ch, 2023-04-02:
This is the Error Popup Box that can not be closed. After pressing RTN the box appears again and again until the transmitter is locked:

[image]

The Text is: "WARNING TX antenna problem"

philmoz, 2023-04-03:
Looking at the code there should be a 10 second delay from when the popup appears to when it will be triggered again.

If you dismiss it quickly does it appear immediately or after about 10 seconds?

icebreaker-ch, 2023-04-04:
- Sometimes, after pressing RTN, the Error Dialog disappears and pops up again after 10 seconds (ok)
- Sometimes, after a few cycles of pop-up/RTN the dialog can not be dismissed by pressing RTN and a long "power off" press is required.
- Sometimes the Transmitter immediately switches off and after powering up again, it falls in "EMERGENCY MODE"
- Sometimes the Transmitter falls directly in "EMERGENCY MODE"

I attach a video, where the transmitter at the end falls in "EMERGENCY MODE":

https://user-images.githubusercontent.com/3645375/229751140-6d69b8b3-d8ce-48c8-81f0-afb542a71866.mp4

philmoz, 2023-04-04:
I can reproduce crashes and EM on my TX16S by opening a popup from the telemetryWakeup function - it looks like it is not safe to create popups from tasks other than the main UI task.

philmoz, 2023-04-04:
@icebreaker-ch are you able to test the build from PR #3439 to see if it fixes the crash issue on your radio.

icebreaker-ch, 2023-04-05:
> 
@philmoz
Sorry, was not able to create a working build (maybe wrong cmake options). I Used the following commands:

```
cmake -Wno-dev -DPCB=X12S -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES  DPPM_UNIT=PERCENT_PREC1 -DCMAKE_BUILD_TYPE=Release ../

make configure

make -C arm-none-eabi -j`nproc` firmware
```

philmoz, 2023-04-05:
You can download the build from the PR in GitHub (via the Actions tab).
For this PR it is here - https://github.com/EdgeTX/edgetx/actions/runs/4612911809
Download the edgetx-firmware-merge zip file at the bottom, unzip it and use the .bin file for your radio.

philmoz, 2023-04-05:
> ```
> cmake -Wno-dev -DPCB=X12S -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES  DPPM_UNIT=PERCENT_PREC1 -DCMAKE_BUILD_TYPE=Release ../
> ```

Take out the -DPCBREV=TX16S and it should work.

icebreaker-ch, 2023-04-05:
Thanks a lot. Your modification fixes my problem. The "TX antenna problem!" box pops up periodically but does not block or bring the transmitter in Emergency Mode.

pfeerick, 2023-04-07:
>The "TX antenna problem!" box pops up periodically

Is that a problem itself, or should it be popping up at that point?

icebreaker-ch, 2023-04-07:
@pfeerick 
This is not a problem itself. It is ok to popup the box periodically, because, selected the external antenna and not have it installed can be a serious problem for your models. So yes, the box shall popup periodically as it is now. The fix from PR #3439 resolves my issue.
@philmoz shall I close this issue?

pfeerick, 2023-04-07:
No, the issue will get closed with the merging of the PR. 

Perfect, thanks for the confirmation. Just wanted to be sure there were no loose ends ;)

PR #3439 (2023-04-16): fix(color) - Fix issue where popup telemetry warning can crash radio.
Fixes #3424 

The antenna warning from the telemetry task is handed off to the UI task for display.

Also fixes an issue where the simulator does not close cleanly if a popup is showing.
closed 2023-04-16 by pfeerick