EdgeTX/edgetx #3425: Double tap to enter app-mode

offer-shmuely, 2023-04-01:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

As we like to promote smarter widgets with app-mode awareness .
For this we need to make the entering/existing app-mode to be as simple & fast as we can.
This will encourage users to use it.

### Describe the solution you'd like

I already added double-tap as exit from app-mode on some of the widgets 
I request to add back the double-tap as it works in 2.7

JimB40, 2023-04-03:
Double with #2629. Switching between apps is PIA without double tap
@raphaelcoeffic is the one who we should be harassing :)

offer-shmuely, 2023-04-03:
sorry, did not find it on the search
closing as duplicated

closed 2023-04-03 by offer-shmuely