EdgeTX/edgetx #3427: MPM 1.3.3.20 some protocols/subtypes not handled in Companion and Firmware Simulator

mha1, 2023-04-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

There is a lot of discrepancies between the MPM protocol list pulled live from the MPM (1.3.3.20) by radio firmware compared to Companion and Firmware Simulator. There are numerous protocols with unspecified subtypes in Companion and protocols that can't be selected using the Firmware Simulator. This is all due to outdated hard coded data in Companion and radio firmware source code.

Using protocol JOYSWAY out of a longer list of affected protocols as an example:

1. Radio with internal MPM 1.3.3.20. JOYSWAY is a valid and selectable protocol with no subtype.
[image]

2. Companion - Protocol JOYSWAY should have no subtypes but shows cryptic subtypes which is the default data Companion uses if there is no hard coded subtype specification found for a given protocol:
[image]

3. Firmware Simulator - Having selected JOYSWAY either in Companion as in 2. and then starting the simulation or using a model file or starting the stand alone firmware simulator using SD model data. JOYSWAY is not recognized at all.
[image]

### Expected Behavior

All MPM 1.3.3.20 protocols/subtypes can be used in a consistent way between Companion, Firmware Simulator and real Radio.

### Steps To Reproduce

Follow example described above.

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S / TX16SMK2

### Anything else?

2.9 recent nightly

pfeerick, 2023-04-03:
This is something we're ever going to really satisfactorily resolve for everyone - as yes, 1.3.3.20 would be right, but what about someone still running 1.3.1.26 (random older number)? This is one reason the radio polls the MPM for the protocols the firmware supports, so only valid protocols are offered. But the list does need to be updated, and idealy made common for both comp and firmware (thus also simu).

mha1, 2023-04-03:
I know, just wanted to show the hard coded MPM protocol data and current MPM offered protocols have already drifted far apart and I don't have a satisfying answer for how to deal with older MPM fw version. Maybe best one can do is to keep up with the at times current MPM fw version, maybe even have CPN/Simulator reading a MPM config file.

Dumb question: does the protocol scan work also for external modules (of all external MPM flavors) or are there situations where the radio is forced to use the hard coded data?

pfeerick, 2023-04-03:
As long as the MPM is running the correct version onwards (1.3.3.0 IIRC),
protocol scan will work. Doesn't matter if its internal or external. Hard
coded list will only be used if earlier firmware, or Jo module present. Its
also possible B&W may do a form of protocol scan with earlier firmware
since the UI works differently - I'm not sure offhand though.

Yeah, some way of using that multi.txt from the MPM releases for both radio
and cpn would be nice.

philmoz, 2023-04-04:
Suggestion:
- Write the MPM version to the radio.yml file on the radio
- Add the Multi.txt files for all the supported MPM versions as Companion resources (named with version #)
- Add the MPM version to the radio settings in Companion (load from radio.yml if present).
- Populate the MPM version dropdown in Companion from the list of available MultiXXX.txt files.
- Load the protocols from the MultiXXX.txt file based on the selected MPM version in Companion.

This makes it somewhat future proof as we only need to add new Multi.txt files to Companion when new MPM versions are created.

elecpower, 2023-04-04:
The list is the easiest part albeit not all protocols are applicable to all modules thus why the dynamic build was introduced on the radios to reduce the available protocols list and query parameters for the chosen protocol.
The complications come from variable number of parameters per protocol, their data type, validation and some special hacks. These can vary between versions.
I started a feature branch a while back with an aim to script the Companion build and had a manual process that got close but the special hacks and oddities prevented taking to full automation.

mha1, 2023-04-04:
There will always be flaws in whatever approach is chosen. For one Multi.txt doesn't have all the information that is required for a fully automatic generation. Information about the individual capabilities is missing, e.g. f/s support, channel map disabled and some very protocol specific things like video frequency etc. And having multiple config files reflecting different MPM firmware versions won't help if you use Companion to edit a new model from scratch. No fw version available so you'd have to take a guess, which most likely will be the most current version. And I personally think it's overkill.

Luckily MPM has matured and there is not a lot of changes to existing protocols to be expected. I'd expect extensions, i.e. new protocols added to an unchanged list of older protocols or the odd new subtype for an existing protocol. This can be handled relatively easy if a new fw version is released. So I'd opt supporting only the latest version as the list of protocols should be backwards compatible. And even if you won't have the latest version and select a protocol the older fw version doesn't support nothing will go up in smoke. The MPM will just not be able to select the protocol. A good reason to upgrade.

So let me jot down my poor mans approach:
- most importantly: have a single source for the Multi protocol definitions
- if there is some new protocols in a new MPM release edit the single source and be done

Please check out my [PR 3435](https://github.com/EdgeTX/edgetx/pull/3435) which focused on creating a consistent data base shared between CPN/Simulator and radio (if scan not possible) code but also fixed some Companion bugs while working on it:
- Concentrated Multi protocol definitions into a consistent single source. Actually two .h files, one for the protocol list and names, the other for subtypes and capabilities, see comments in .h files as to why
- updated available protocols reflecting 1.3.3.20 (15+ protocols) 
- fixed Companion always showing f/s and channel map options in model setup disregarding the protocol capability

closed 2023-05-22 by pfeerick