EdgeTX/edgetx #3428: Scroll wheel on an X12S is difficult to operate

icebreaker-ch, 2023-04-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The scroll wheel of my FrSky X12S is very difficult to operate. For example, when trying to set numerical values or points of a curve, the underlying value often jumps up and down to the MIN or MAX value. I really have to turn the scroll wheel VERY slow, to achieve a certain value. This can be annoying, if the range is quite big. I never had problems that way on the previously installed OpenTX system, so this is not a hardware problem in my radio.

### Describe the solution you'd like

An "OpenTX" behaviour of the X12S scroll wheel.

### Describe alternatives you've considered

Turn the scroll wheel very slow (can take "forever" to achieve the wanted value)

elekSENDER, 2023-04-03:
> the underlying value often jumps up and down to the MIN or MAX value.

I can confirm this behavior as well on a X10s.
Firstful i thought of an issue of my rotary encoder, but you described exactly the same as occured on my X10s.
Maybe there is any connection and it can be solved together with #3409.

philmoz, 2023-04-04:
Does it jump based on the direction of rotation (i.e. if you are increasing the value does it jump to MAX)?

If you go to the SYS -> Hardware -> Debug -> Keys page there is an 'R.E.' entry at the bottom.
Does this increase and decrease correctly as you rotate the dial?

icebreaker-ch, 2023-04-04:
1. Yes it jumps on the direction of rotation
2. In the SYS -> Hardware -> Debug -> Keys page, the R.E. entry does increase or decrease absoluteley correctly on rotation

I attach a video of the bad behaviour:
https://user-images.githubusercontent.com/3645375/229742995-07e434c2-53ee-4239-b8b4-e0e86198f354.mp4

elekSENDER, 2023-04-04:
> 1. Yes it jumps on the direction of rotation
> 2. In the SYS -> Hardware -> Debug -> Keys page, the R.E. entry does increase or decrease absoluteley correctly on rotation

Confirmed the same on a X10s.
I recognized this behavior the first time at mixes weight and it jumps to the corresponding rotation direction between -500 an +500%.

philmoz, 2023-04-05:
It looks like there is an issue with the way acceleration is being handled for the rotary encoder on color LCD radios.

The code calculates an acceleration factor based on, the time between successive rotations in the same direction, and how many steps the dial has been turned.

This is then used by the number edit control (and a few others) to increase the rate of change.

However the code is also applying the acceleration factor to the number of steps the dial was turned.
So instead of 4 steps with an acceleration of 10 you get 40 steps sent to the number editor which also then applies the acceleration and the value jumps too far.

philmoz, 2023-04-06:
Can you please test the build from PR #3441 and see if helps.
Build can be downloaded from - https://github.com/EdgeTX/edgetx/actions/runs/4625160889

elekSENDER, 2023-04-06:
On my X10s this build works perfect with the wheel, really a lot better in all tried menus.

icebreaker-ch, 2023-04-06:
The handling is MUCH better now, although I would prefer even less acceleration on the scroll wheel input for an X12S, because the scroll wheel on the X12S is not really good to handle (mechanically). The scroll wheel of the X10 definitely has a better haptic.

philmoz, 2023-04-06:
> The handling is MUCH better now, although I would prefer even less acceleration on the scroll wheel input for an X12S

Is it just on number edit where it needs less?

icebreaker-ch, 2023-04-06:
I mainly tried on moving curve points (which is in fact is also number editing). Navigating in the UI seems ok for me.
Have to make further tests on my X12S and also on the X10S. Maybe I can propose a piece of code that makes me happy...

icebreaker-ch, 2023-04-06:
To make acceleration less aggressive, I tried this on line 275 in LvglWrapper.cpp:

 auto dx_dt = (diff * diff / 2 * 100) / dt;

feels goodf to me.

Luznatural, 2023-04-07:
I have the same problem. This solution (or one with similar effect) is going to be incorporated as standard in follow-ones firmwares?

theramman, 2023-04-13:
Yeah, I finally figured that one out. the cmake string I'm using is the following: 
cmake -Wno-dev -DPCB=X12S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DCMAKE_BUILD_TYPE=Release ../
And it only compiles a 471k firmware image. I know that's not right. The standard v2.8.2 image is 1.5mb.... What am I missing?

pfeerick, 2023-04-13:
If you didn't clear the cmake cache, it's probably trying to build some frankenstein version of the an x9d firmware... 

It would be simpler to simply do the following at the github prompt (using the same build script as used for github firmware builds):
```
cd ..
FLAVOR=X12S tools/build-gh.sh
```

PR #3441 (2023-04-16): fix(color): Rework rotary encoder acceleration calculation removing double handling
Fixes #3428 

Only tested on TX16S Mk2 so far; but this seems to work better in my testing.
closed 2023-04-16 by pfeerick