EdgeTX/edgetx #3434: Lost FUTABA SFHSS protocol ?

LS-67, 2023-04-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The receivers of models under ETX 2.8.1 and FUTABA SFHSS protocol use automatically  a "J6" protocol under ETX 2.8.2 and are no longer bindable.
When I create a new model with the FUTABA SFHSS protocol under ETX 2.8.2, this protocol disappears again after switching off/on the TX, to become "J6"

### Expected Behavior

keep the SFHSS protocol like in ETX 2.8.1

### Steps To Reproduce

1. install ETX 2.8.2
2. open a model that was previously in SFHSS
3. select "Model settings
4. select "internal RF"

### Version

2.8.2

### Transmitter

Radiomaster TX16S / TX16SMK2

coaster007, 2023-04-04:
similar problem with radiomaster boxer 2.8.2 (External MPM)

Selected protocol Eachine E129. 
After off/on Radio the protocol changed to JOYSWAY.
Reverted to 2.8.1 everything works again.

mha1, 2023-04-04:
You can try this tx16s firmware based on current 2.8.2 built with PR 3435 on top of it. I'd appreciate your feedback if this fixes your problem too. 
[tx16s-2.8.2-PR Multi Single Source.zip](https://github.com/EdgeTX/edgetx/files/11151417/tx16s-2.8.2-PR.Multi.Single.Source.zip)

mha1, 2023-04-05:
try this please [tx16s-2.8.2-PR Multi Single Source1.zip](https://github.com/EdgeTX/edgetx/files/11158074/tx16s-2.8.2-PR.Multi.Single.Source1.zip)

LS-67, 2023-04-05:
it still does not work !

mha1, 2023-04-05:
What exactly are you doing? You are doing everything on the radio, no Companion, correct?

I tried this but can't reproduce your problem:
- TX16s, flashed 2.8.2 release, Multi 1.3.3.20
- power up radio
- go to select model
- hit new model, select blank model
- go to model settings
- hit internal rf, select Multi, select Futaba (SFHSS comes up as RF Protocol)
- turn off radio
- turn on radio
- failsafe warning pops up, press RTN to continue
- go to model settings
- select internal rf
- Futaba SFHSS is still selected

Please post exact procedure to reproduce

mha1, 2023-04-05:
This is odd. Where did you get the firmware? self build? language build? 

Go to radio setting, version. Please post a screen shot of the internal RF / external RF settings
Go to radio setting, version. Please post a screen shot.

go to model setup, internal RF (does ist say 1.3.3.20?)
hit Mulit and set to Off
turn of radio
turn on radio
go to model setup, internal RF
change from Off to Multi again
does a pop come up saying scanning (or similar)

LS-67, 2023-04-05:
The firmware is tx16s-2.8.2-selfbuild (9c3c6c3e)
Date 2023-04-05
Time 11:40:19
OPTS : bluetooth, crossfire,ghost,afhds3,noheli,internalgps,internalmilti,internalaccess,multimodule,luac,cli,passthrough,cli
In the model setup, internal RF, the module status is V1.3.3.20 TAER
When i do this : 
"hit Mulit and set to Off
turn of radio
turn on radio
go to model setup, internal RF
change from Off to Multi again
does a pop come up saying scanning (or similar)"

Yes, a pop come up saying "MPM Scanning protocols"

mha1, 2023-04-05:
I doubt it makes a difference but please use the official release and try the sequence I first posted again. 
[tx16s-03e80fb-2.8.2-release.zip](https://github.com/EdgeTX/edgetx/files/11160857/tx16s-03e80fb-2.8.2-release.zip)

coaster007, 2023-04-05:
@mha1 the 2.8.2 version is downloaded from the edgetx homepage (official release).
I haven't see any "MPM scanning  ..." screen.
the question is why the 2.8.2 version changed the index in the multiprotocol.txt file to +1 ?

mha1, 2023-04-05:
@coaster007 there was a change for some invaldi protocol (XK). But so far I can't reproduce your problems. I wonder what is different.

@LS-67 have you tried this:
```
    TX16s, flashed 2.8.2 release, Multi 1.3.3.20
    power up radio
    go to select model
    hit new model, select blank model
    go to model settings
    hit internal rf, select Multi, select Futaba (SFHSS comes up as RF Protocol)
    turn off radio
    turn on radio
    failsafe warning pops up, press RTN to continue
    go to model settings
    select internal rf
    Futaba SFHSS is still selected
```

please pot your your exact procedure to reproduce

LS-67, 2023-04-05:
I do this "TX16s, flashed 2.8.2 release, Multi 1.3.3.20
    power up radio
    go to select model
    hit new model, select blank model
    go to model settings
    hit internal rf, select Multi, select Futaba (SFHSS comes up as RF Protocol)
    turn off radio
    turn on radio
    failsafe warning pops up, press RTN to continue
    go to model settings
    select internal rf
    Futaba SFHSS is still selected"

Now, it's OK for a blank model (v2.8.2"FlyingDutchman") (even after turn off/on TX)

But still not OK with a model which was SFHSS before with ETX 2.8.1!

coaster007, 2023-04-05:
o.k. the same test here (radiomaster boxer external MPM) new model with protocol E129 power off/on protocol still E129.
But existing E129 protocol changed to JOYSWAY

mha1, 2023-04-05:
please post the 2.8.1 model.yml file of the model with Futaba/SFHSS selected

LS-67, 2023-04-05:
How should I do to post it ? Drag and drop doesn't support the file type yml ?

mha1, 2023-04-05:
zip it?

mha1, 2023-04-05:
Thanks, this looks ok. If you can, please post the yml file showing the wrong protocol in 2.8.2

LS-67, 2023-04-05:
Here it is
[NewFOKONG2.zip](https://github.com/EdgeTX/edgetx/files/11161355/NewFOKONG2.zip)

mha1, 2023-04-05:
Well, at least it confirms what you are seeing:
moduleData:

before
```
  0:
    type: TYPE_MULTIMODULE
    subType: 21,0
```

after
```
  0:
    type: TYPE_MULTIMODULE
    subType: 22,0
```
means:
21,Futaba,SFHSS
22,J6PRO

I can reproduce your problem in the simulator. Let me see if I can find the root cause

LS-67, 2023-04-05:
Thank's for your help !

mha1, 2023-04-06:
please try this tx16s 2.8.2+ firmware and check if your problem still persists: [tx16s-2.8.2+.zip](https://github.com/EdgeTX/edgetx/files/11169958/tx16s-2.8.2%2B.zip)

LS-67, 2023-04-06:
It seems to work...perfectly ! Thank you for the time you spent to solve my problem. (vers 2.82-selfbuild 03e80fb7)

PR #3444 (2023-04-13): fix: undesired MPM protocol selection changes
Fixes #3434

Summary of changes:
This fixes the shortcomings of [PR #3355](https://github.com/EdgeTX/edgetx/pull/3355) where ACCESS receiver name data was corrected to fix https://github.com/EdgeTX/edgetx/issues/3273
Added checking module types of the currently loaded model. If there is no PXX2 module selected there is nothing to correct. 
The problem finder confirmed this to fix his problem.

Note of shame:
Dummy me created a major mess with https://github.com/EdgeTX/edgetx/pull/3355 which showed it's ugly face in 2.8.2 (https://github.com/EdgeTX/ [... cut]
closed 2023-04-13 by pfeerick