EdgeTX/edgetx #3438: Splash screen and welcome audio

audio98, 2023-04-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No

### Describe the solution you'd like

Is it possible to have the welcome splash screen and welcome audio come on at the same time and last for 7 seconds. Similar to OpenTX. Thanks

Kevltan, 2023-04-07:
Agree with that 👍👍

philmoz, 2023-04-08:
PR #3451 addresses this; however it is not possible to start playing the welcome sound immediately when the splash screen loads. This is due to other initialisation that needs to happen before the startup sound plays.

PR #3451 (2023-07-30): feat(radio,cpn): Startup splash screen and sound changes.
Fixes #3438, fixes #2537, fixes #391 

Summary of changes:
- Enable startup splash screen duration for color LCD radios.
- Add option to disable startup sound on all radios.
- Add Companion support for above changes.
- Move mixer task startup to after splash screen is done on color LCD radios so that the startup sound is played first.

Notes:
- The startup splash screen cannot be disabled for color LCD radios. The minimum setting is 1 second. This is because the radio.yml file has not been loaded when the splash screen is displayed so t [... cut]
closed 2023-07-30 by pfeerick