EdgeTX/edgetx #3440: two drives come up when USB is plugged into radiomaster

ror1948, 2023-04-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

two drives come up on PC when USB is plugged into radiomaster in storage mode.
one drive is sdcard contents
second drive is firmware drive.
i thought their was a nightly build a long time ago that only allowed the SDCARD drive to come up.
I had loaded the nightly build and it actually only had one drive come up.
Why is that feature not in 2.8.2 ?
thanks

### Expected Behavior

when USB is plugged into radiomaster on a PC.
only one drive of sdcard contents pops up on PC

### Steps To Reproduce

Tx on
connect USB to PC
Insert USB into top USB port.

### Version

2.8.2

### Transmitter

Radiomaster TX16S / TX16SMK2

### Operating System (OS)

Windows

ror1948, 2023-04-05:
ok the above seems to be fixed in nightly tx16s-50b492d but was not included in the 2.8.2 release.
Am i missing something or misunderstanding how this all works.
Thanks

pfeerick, 2023-04-05:
Removal of the firmware drive is a new "feature" for 2.9, whereas any 2.8.x (and indeed 2.7.x and 2.9.x) releases are primarily maintenance and bugfix releases, not new features.

closed 2023-04-05 by pfeerick

ror1948, 2023-04-05:
So nightly builds have all current fixes and features but releases are selective?

- Ralph

pfeerick, 2023-04-07:
Nighties are nightly builds of the current development branch, which is working towards the next major version. Once a version is branched off the main development line, i.e. 2.8, there will be no new major features added to it, primarily only bugfixes. 

So yes, a nightly is all the current fixes and features, whereas a release is a snapshot in time.