EdgeTX/edgetx #3450: Channel monitor graphical/display glitches

bigtwin, 2023-04-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The problem is already reported/fixed with OTX 2.3.9  [Channel monitor graphical/display glitches](https://github.com/opentx/opentx/issues/7967)

I just upgraded my _Taranis x9lite_ (Access) to ETX 2.8.2 and experienced the same. I cannot say whether the already existed with ETX 2.7.1.

### Expected Behavior

Properly displayed channel monitor

### Steps To Reproduce

Simply "page" main screen to the channel monitor

### Version

2.8.2

### Transmitter

FrSky X9 Lite / Lite S

### Operating System (OS)

macOS

pfeerick, 2023-04-08:
The fix that was applied there is present in ETX, so perhaps it needs to be expanded a little wider to not just be limited to the X7 and X9LITES
https://github.com/EdgeTX/edgetx/blob/6dd63cb8e09944966836ca917fdf0268ecfcb606/radio/src/gui/128x64/widgets.cpp#L71-L76

pfeerick, 2023-04-09:
Are you able to flash this firmware, and let me know if that fixes the issue for you?
[x9lite-2.8.2-test.zip](https://github.com/EdgeTX/edgetx/files/11185236/x9lite-2.8.2-test.zip)

```diff
diff --git a/radio/src/gui/128x64/widgets.cpp b/radio/src/gui/128x64/widgets.cpp
index 663ca73f5..e30a17c29 100644
--- a/radio/src/gui/128x64/widgets.cpp
+++ b/radio/src/gui/128x64/widgets.cpp
@@ -68,7 +68,7 @@ void drawVerticalScrollbar(coord_t x, coord_t y, coord_t h, uint16_t offset, uin
 
 void drawGauge(coord_t x, coord_t y, coord_t w, coord_t h, int32_t val, int32_t max)
 {
-#if defined(PCBX7) || defined(PCBX9LITES) // X7 LCD doesn't like too many horizontal lines
+#if defined(PCBX7) || defined(PCBX9LITE) || defined(PCBX9LITES) // X7 LCD doesn't like too many horizontal lines
   h++;
 #else
   lcdDrawRect(x, y, w+1, h);
```

bigtwin, 2023-04-09:
Hello @pfeerick!
Thx for taking care that fast! I could test it right away and can confirm - problem resolved :-)

pfeerick, 2023-04-10:
No problem, thanks for the quick test. A fix will be present in 2.8.3 and 2.9.x. You should be able to safely use the firmware I sent you if you wish as it is 2.8.2 with some 2.8.3 fixes applied.

richardclli, 2023-04-10:
> How is your preferred workflow? Should this issue remain open till 2.8.3 is released, will you close it or am I supposed to close the issue since I started it? I am supporting a collegue who is working on a german translation of the ETX manual and I am depending on a german install. Therefor I will roll back to official 2.8.2 and patiently waiting for 2.8.3 :-)

You just leave the closing of an issue to us, we will manage it thanks.

PR #3452 (2023-04-13): fix(X9lite): Chan. monitor, other display glitches
It seems the display used on these TX doesn't like too many horizontal lines.

Fixes #3450

Summary of changes:
 - Apply X7/X9 Lite S fix to X9 Lite also, consistently
closed 2023-04-13 by pfeerick