EdgeTX/edgetx #3455: Zorro Splash Screen brightness/contrast

mixMugz, 2023-04-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have a custom Splash Screen in my Zorro TX. When I turn TX on, I see my splash, but it's very bright, and seems like the contrast is set to a minimum. If I set the contrast in TX to the maximum, my splash screen keeps the same (low bright) behavior.

How I can control my contrast values for boot splash? Why it not used TX values?

Look second screenshot.

[image]
[image]
[image]

### Expected Behavior

The splash screen uses the main contrast value of TX.

### Steps To Reproduce

1. Set a custom splash screen with lots of blacks
2. Set contrast in TX to maximum value
3. Turn off TX
4. Turn back TX to ON
5. Look the splash screen

### Version

2.8.2

### Transmitter

Radiomaster Zorro

### Operating System (OS)

Windows

### OS Version

Window 10 22H2

pfeerick, 2023-04-10:
Contrast value will be the firmware default, as at the time the splash screen is shown, the SD card hasn't even been mounted yet.

mixMugz, 2023-04-10:
> Contrast value will be the firmware default, as at the time the splash screen is shown, the SD card hasn't even been mounted yet.

How do I change the default value of firmware (if I wish to build custom firmware)? For now, it seems like the default value is set to a minimum.

mixMugz, 2023-04-10:
I think it will be a good idea if the default value will be middle between min and max.

pfeerick, 2023-04-10:
It basically is, but it tweaked slightly for different radio screens. For the zorro, it's probably set to 15 (where the range is 10 to 30). 

If you're up for building a custom firmware, try adding ` || defined(RADIO_ZORRO)` to line 848 and see if that is enough... maybe it just needs to be lifted just that little bit more. 

https://github.com/EdgeTX/edgetx/blob/745c465bedbaea3657bfd453c6e60469be775ba8/radio/src/targets/taranis/board.h#L834-L859

mixMugz, 2023-04-10:
> It basically is, but it tweaked slightly for different radio screens. For the zorro, it's probably set to 15 (where the range is 10 to 30).
>
> If you're up for building a custom firmware, try adding ` [...]

So, this workaround worked great! Now the Splash screen looks like all other screens, but now I have a boot problem. It's too dark =( There is any place in the code where I can set contact for the initial boot (dots)??

[image]
[image]
[image]
[image]

closed 2023-05-09 by pfeerick

pfeerick, 2023-05-09:
> There is any place in the code where I can set contact for the initial boot (dots)??

Unfortunately, not anything that has been specifically designed for that... you'd probably have to fudge with the contrast in the startup animation function to make that look nicer. As good as the B&W screens for normal usage, they also seem somewhat ... cheap and nasty for want of a better description. While there seems to be something a bit weird about how we use them as the "contrast" changes depending on how much content there is on the screen, to be honest, I've not looked at the B&W lcd driving code so it may just be a function of how these screens work.

PR #3575 (2023-05-09): Added default contrast settings for Zorro
Fixes #3455

Summary of changes:
Just added default contrast settings for Radiomaster Zorro