EdgeTX/edgetx #3457: TX16S Mark II : USB not working on 2.8.1 and newer

zwieblum, 2023-04-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

USB storage/joystick/serial does not work in 2.8.1 and newer.  The kernel log shows that the device enumeration failes, so the radio does not detect the PC and does not ask for action (storage, joystick, serial). The radio config for USB is set to "ask"

### Expected Behavior

2.8.0: Power radio on, plug in USB --> I get a dialog on the radio that asks for action (storage, joystick, serial). The selected action works as expected.

### Steps To Reproduce

1) power radio on
2) connect to PC
3) nothing happens on 2.8.1 and newer, "action" dialog pops up on 2.8.0

### Version

2.8.2

### Transmitter

Radiomaster TX16S / TX16SMK2

gagarinlg, 2023-04-10:
Which operating system are you using?
I used my TX16S with 2.8.2 the last couple of days, multiple times per day without issues. I used Windows 10.

zwieblum, 2023-04-10:
I run Linux devuan daedalus with kernel 6.1.15-1 on AMD Ryzen. I trued Windows 10 on the same platform, but that did not even give a mesage that something is wrong.
I also tried on an old X61 with devuan beowulf kernel 4.19.152-rtai-amd64 - same effect als on newer kernel, enumeration failes.

philmoz, 2023-04-10:
Can you provide the radio.yml and modelXX.yml files please.

richardclli, 2023-04-11:
I do not have any problem with my TX16S mark I

rotorman, 2023-04-11:
The bottom USB port on TX16S (first gen and mkII) is solely for charging purposes. No data can be transferred via bottom USB-C port, it is not connected to STM32 microcontroller.

zwieblum, 2023-04-11:
Not connected sounds strange to me. IMO I should not get a enumeration error in that case, should I?

rotorman, 2023-04-11:
The bottom USB-C port of TX16S mkII is connected to the charge controller chip that works independently of the STM32 microcontroller, where EdgeTX is running.

gagarinlg, 2023-04-11:
The bottom port is only connected to the charging IC

zwieblum, 2023-04-11:
Oh well ... thanks you for the explanation ... and sorry about the fuss. Maybe it would be a good idea to share the info on the radio page?

richardclli, 2023-04-13:
So this is resolved.

closed 2023-04-13 by richardclli