EdgeTX/edgetx #3458: Companion and simulator (X9D+) crashes changing external multi module protocol type

elecpower, 2023-04-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

Companion and simulator crashes when scrolling through the external multi module protocol types using +/- keys or mouse scroll wheel.
At the time of posting the bug has only been found with X9D+ radio simulator.

### Expected Behavior

No crash

### Steps To Reproduce

Companion using X9D+ profile
1. Add new file
2. Add new model
3. Edit model
4. Select Setup tab
5. Go to External Radio Module
6. Set protocol to Multi
7. Close edit window and click Simulate Radio

X9D+ simulator
8. Click MENU
9. Click PAGE
10. Go to External RF sub field Type

[image]

11. Click ENT to enter edit mode
12. Using mouse scroll up/down OR click +/-
13. After a varying number of mouse scroll up and down OR click +/- and Companion and simulator crash. The simulator ui stops responding before the crash.
 
The speed of change does not appear to affect the speed of crash. In some tests it has not crashed though not crashing is rare.

If run from a terminal session a Segmentation Fault message is displayed.

### Version

2.8.2

### Transmitter

FrSky X9D+

### Operating System (OS)

Linux, Windows

### OS Version

Ubuntu 20.04, Win 10

### Anything else?

The bug has been found in official 2.8.1, 2.8.2 and self build main and PR 3435 branches.
Boxer simulator appears from testing to be unaffected.
Confirmed with assistance of @mha1 

Additional simulators to be tested to gauge the extent of radios affected.

Would be great if someone could test on hardware.

elecpower, 2023-04-10:
Further testing of B&W simulators
Crash: X9D+, X9D+ 2019, X9D, X9E
Pass: X7 / X7S, X7 / X7 ACCESS, X9 Lite, TX12 Mk II
So at this time it appears limited to 212 x 64 LCD

elecpower, 2023-04-10:
After the ui stops responding it also takes a few seconds to crash and varies depending on whether I'm running in Win 10 or Ubuntu VM. Win 10 takes longer than Ubuntu and with Win 10 having more memory it would appear it is a loop that is allocating memory and eventually runs out.

philmoz, 2023-04-10:
Missing line:
             if (!title) break;
in model_setup.cpp, ITEM_MODEL_SETUP_EXTERNAL_MODULE_OPTIONS case (line 1546).

PR #3459 (2023-04-11): fix(radio) - Fix crash when editing multi module type on 212x64 LCD B&W radios.
Fixes #3458 

Add missing test for empty 'title' value.
closed 2023-04-11 by pfeerick